Ultimaker/Cura #4448: Crashes when browsing packages

hugoabernier, 2018-09-30:
**Application Version**
3.4.1 and 3.5.0

**Platform**

**Printer**
None

**Steps to Reproduce**
Click on Preferences and select Browse Package. 

**Actual Results**
Cura will hang

**Expected results**
Should see a list of packages

**Additional Information**
Running in Windows 10

fieldOfView, 2018-09-30:
You mean _Toolbox_, then Browse Packages, right? Because neither 3.4 not 3.5 have a Browse Packages under Preferences.

For crashes and similar issues, please attach the following information:

    (On Windows) The log as produced by dxdiag (start -> run -> dxdiag -> save output)
    The Cura GUI log file, located at
        %APPDATA%\cura\<Cura version>\cura.log (Windows), or usually C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log
        $USER/Library/Application Support/cura/<Cura version>/cura.log (OSX)
        $USER/.local/share/cura/<Cura version>/cura.log (Ubuntu/Linux)

If the Cura user interface still starts, you can also reach this directory from the application menu in Help -> Show settings folder

diegopradogesto, 2018-10-02:
Please provide the logs, as @fieldOfView indicated. Does it happen in both 3.4.1 and 3.5 versions?

MartinBienz, 2018-10-06:
This happend to me as well. It migrated the octoprint plugin to the new version (in my case 3.5). Then the browser would log an error, that he had a problem with an old octoprint plugin. So, I wen to %APPDATA%\cura\3.5, checked the "plugin" folder and deleted all content (you can also rename and create a new folder). After that it would work.

diegopradogesto, 2018-10-17:
@MartinBienz I wouldn't recommend to manually delete the plugins, since we have an internal file to keep track of all the packages that were installed. If you delete it manually, the file won't be updated anymore. In that case you can see weird states afterwards (such as seeing a plugin as installed when it's not, or maybe seeing the update button when the plugin is not installed). I recommend you to always use the _Installed_ tab in the Toolbox.

@hugoabernier do you still have this problem? Please give us more information.

MartinBienz, 2018-10-17:
@diegopradogesto Yeah that's fine, so what's the proper way of doing this then? I mean if Cura's Toolbox ist not launching / hanging... how would one then fix it? Are not only the 3rd party plugins stored in that %appdata% folde (like octoprint etc.). Should not be an issue to delete that folder an start fresh (my opinion). What / where is the "internal" file then, can we not just reset it to "default", delete all plugins and start from scratch, should be ok, no?

diegopradogesto, 2018-10-17:
The file is called "packages.json" in the root of the configuration folder. If you want to clean all your plugins from your folder by manually deleting the "plugins" folder, then I recommend you to edit this file. By default (when it's empty) it looks like this:

```
{
    "installed": {},
    "to_install": {},
    "to_remove": [],
    "version": 1
}
```

diegopradogesto, 2019-09-11:
Does this fixed your problems? Are you still experiencing that Cura hangs when open the (now called) Marketplace?

MartinBienz, 2019-09-11:
@diegopradogesto yes, problems are gone. Never had the same issue again, even after updating multiple times.

closed 2019-09-12 by nallath