Ultimaker/Cura #4449: Print Setup tabs disappear

UUeekend-Design, 2018-09-30:
**Application Version**
3.4.1

**Platform**
win10 x64

**Printer**
custom printer, based on makerbot

**Steps to Reproduce**
minimizing one tab on the printer setup area. It could also be that I had this settings:
wall line width             0
outer wall line width   0,35
inner wall line width    0,4
the lower two was manuallhy set by me so the top one was greyed out, fooling me to believe it would work. However, it wouldn't slice the part for me so I removed that and reentered 'wall line width 0,4 and then it would let me slice.. could this trigger the problem you think?

**Actual Results**
all tabs disappear, since it thinks they are all minimized, the scroll bar to the right is also removed. But 98% of the times it works, so I dunno how you can test this. I also don't know how to save a log file.. :/

**Expected results**
minimizing one tab should conly minimize that specific tab and leave it in the menu in case i would like to maximize it to Review my settings again

**Additional Information**
I had this a couple of weeks ago but since I'm new to these different slicers I just rage quitted and tried another one instead. I didn't pay to much attention to why, but now, I got this working rather good on occasion and I'd like it to never expose me to this annoying problemever again..

diegopradogesto, 2018-10-02:
Following your steps, I don't see any tab/category disappearing. If it happens often to you, please provide a project file (in 3.4.1 go to _File->Save Project_).

You can find the logs in the configuration folder (_Help->Show configuration folder_). It's a file called **cura.log**. You can also zip this whole 3.4 folder and send it to us, so that we can completely reproduce the issue with your exact environment.

Ghostkeeper, 2018-11-23:
I can also not reproduce this. Maybe it's now no longer applicable (or in the new UI anyway). If you find this again, let us know.

Here's some instructions on where to find the log file: https://github.com/Ultimaker/Cura#logging-issues

closed 2018-11-23 by Ghostkeeper