Ultimaker/Cura #4451: Insane Printing times

gsohler, 2018-10-01:
I used to print with PLA only. But when i tried to switch to NYLON,
i must have messed up my settings, even If i move back to PLA.
For a small piece, printing time is 4 days, and slicing does not reliable finish in (less than 30 seconds !!!)
I checked the relevant settings like travel speed, layer height, object dimension. even printing simulation
appears normal behaviour.
Is there anything else i can check or even reset to facrtory settings ?

[image]

This is the layer view, which appears to have strange home posiitons but when i check the machine and extruder settings, everything appears normal.

what could be wrong here ?

**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

fieldOfView, 2018-10-01:
> Is there anything else i can check or even reset to facrtory settings ?

Next to "Draft Quality #2", there is a star. Click it, and you will be brought into the Profiles pane of the preferences, with the current profile selected. There are tabs that show you the settings in "Draft Quality #2" but also "user" settings. Those are the changes that you made to the profile. Check both the global and extruder tab. There's also a button to reset to Draft Quality 2, discarding any changes to that profile you have made since creating the profile.

gsohler, 2018-10-01:
Hi and thank you for your hints.
I was able to see my personal changes and all of them made sense. Then I reset them. 
After pressing "prepare" printing time is 17 hours.  I moved my model to the left and pressed "prepare" 
again. Now its stuck in "slicing" unless i press "Cancel"
Next I uninstalled Cura and installed it again, but my settings are still there.  How can i delete all my setup and start from scratch ?

Günther

fieldOfView, 2018-10-01:
Go to Help -> Show configuration folder.
Close Cura
Remove all files from the configuration folder (but leave the folder in place)
Restart Cura

gsohler, 2018-10-01:
It appears, that this issue only happens with dedicated STL file. The STL file works fine with the old Cura 15.41.
For license reasons i cannot disclose it this is why i close the issue.

Thank you for your support

gsohler, 2018-10-02:
its the files from 

https://www.thingiverse.com/thing:3106763
(Part2.stl)
This worked fine for the old Cura 14.51 but with recent cura, it fails

Vandrasc, 2018-10-02:
Hi @gsohler , is it possible to share the project file with us? File>Save.. 
I am not able to reproduce, they may be some settings which are causing the extra moves, the printing time for me with the part2.stl is 1h. 

Which recent Cura version are you using? 
To simulate a clean install make sure to remove both folders of Cura from User/AppData/Roaming and Local.

diegopradogesto, 2018-10-02:
Is this a duplicate of #4009?

gsohler, 2018-10-02:
Hi Vandrasc,

Thank you for your hints.
I deleted all cura folders in Local and Roaming.
When starting cura again and setting up  my printer again,
the problem is gone.
So the Problem was with my settings, not with Cura itself.
Its a pity, i did not do File->Save(my intent was to do after it still created strange printing times)
I am using Cura 3.4.1
I can close the issue again. Thank you for your efforts

closed 2018-10-02 by gsohler