Ultimaker/Cura #4454: Gcode start with M190 , M104 , M109 , M82

RDS356, 2018-10-01:
My 3d make the g28 and g29(auto leveling) before M109 and M190.
Why ? I have a great system of Zprobe thats works very well with piezoeletric system and I need make the zprobe before hotting nozzle.
But the new version of cura 3.4.1 ever put this M109 and M190  first of my inicial gcode script.
Is possible make like olders versions when the inicial g-code scipts appers every first?

This is my inicial gcode script;
G90        ;absolute positioning rds
M82        ;set extruder to absolute mode
M107       ;start with the fan off
G28 ;move all to min endstops
G29     ;auto leveling
M190 S{print_bed_temperature} 
M109 S{print_temperature}
G1 X100 Y5 Z10.0 F6000 ;move para limpeza
G4 P500
M300 S800 P2500
G4 P500
G4 P6000  ;pausa 6 segundos
G92 E0                  ;zero the extruded length
G1 F200 E3              ;extrude 3mm of feed stock
G92 E0                  ;zero the extruded length again
G1 F{travel_speed}
;Put printing message on LCD screen
M117 Imprimindo...

fieldOfView, 2018-10-01:
Cura inserts a heating sequence because you are using deprecated replacement tags. Use `material_print_temperature` instead of `print_temperature` and `material_bed_temperature` instead of `print_bed_temperature`.

The M82 cannot easily be prevented, but it should not matter if it is output before or after your zprobing. If you really want to get rid of it, you can use a postprocessing script.

PS: `travel_speed` won't work, use `speed_travel` instead.

RDS356, 2018-10-02:
Hi fieldOfView..
Worked fine..
Thank you too much.
Best Regards,
Reinaldo de Souza - From Brazil.

fieldOfView, 2018-10-02:
Can you close the issue?

closed 2018-10-02 by RDS356