Ultimaker/Cura #4456: Unable to slice on Cura 3.4.1

Gritz009, 2018-10-01:
**Application Version**
Cura 3.4.1

**Platform**
Windows 10 (64)

**Printer**

[Creality ender 3.curaproject.zip](https://github.com/Ultimaker/Cura/files/2434590/Creality.ender.3.curaproject.zip)

**Actual Results**

[image]

diegopradogesto, 2018-10-02:
Same project file slices correctly for me in the same 3.4.1 version and Windows 10. Does it happens every time? Can you please provide some reproduction steps?

fieldOfView, 2018-10-02:
Logs are needed to be sure, but it could be that an overzealous virus checker is blocking communication between Cura and CuraEngine.

Gritz009, 2018-10-02:
so how do i fix it....this is working fine on my friends computer which has win 8.
But, I am unable to do anything.
please suggest a fix

diegopradogesto, 2018-10-02:
Send the logs and try what @fieldOfView suggests: disable the virus checker or even the firewall, to see if something is blocking communication between Cura and CuraEngine.

Gritz009, 2018-10-03:
I tried the common steps like running as admin, disabling antivrus 
i am attaching the log file. please suggest an appropriate solution
[cura.log](https://github.com/Ultimaker/Cura/files/2441844/cura.log)

fieldOfView, 2018-10-03:
This is the interesting bit from the logs:
```
2018-10-03 13:38:33,793 - ERROR - [MainThread] UM.Logger.logException [81]: Exception: Unable to find backend executable: C:\Program Files\Ultimaker Cura 3.4\CuraEngine.exe
2018-10-03 13:38:33,795 - ERROR - [MainThread] UM.Logger.logException [85]: Traceback (most recent call last):
2018-10-03 13:38:33,796 - ERROR - [MainThread] UM.Logger.logException [85]:   File "X:\3.4\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 143, in _runEngineProcess
2018-10-03 13:38:33,798 - ERROR - [MainThread] UM.Logger.logException [85]:   File "subprocess.py", line 911, in __init__
2018-10-03 13:38:33,799 - ERROR - [MainThread] UM.Logger.logException [85]:   File "subprocess.py", line 1126, in _get_handles
2018-10-03 13:38:33,800 - ERROR - [MainThread] UM.Logger.logException [85]:   File "subprocess.py", line 1006, in _get_devnull
2018-10-03 13:38:33,801 - ERROR - [MainThread] UM.Logger.logException [85]: FileNotFoundError: [Errno 2] No such file or directory: 'nul'
```

Apparently, it's possible for the service implementing the null device ('nul' pseudo-file) to be stopped or to malfunction.

The status can be checked via the control panel. Start the Device Manager (Control Panel > Hardware and Sound > Device Manager), select View > Show hidden devices, then under Non-Plug and Play Drivers, there should be an entry for "Null".

The corresponding registry key is:
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\services\Null
The Start and Type values should both be 1. `

Gritz009, 2018-10-03:
I do not find Non-Plug and Play Drivers in my device manager

Gritz009, 2018-10-03:
@fieldOfView i checked my registry 
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\services\Null
The Start and Type values both are 1

the problem is solved...THANK YOU VERY MUCH..!!!!!

closed 2018-10-03 by Gritz009