Ultimaker/Cura #4458: [master]  translation failure

smartavionics, 2018-10-02:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

````
/home/markb/3dp/cura/Cura.git/resources/i18n/pl_PL/fdmprinter.def.json.po:10: end-of-line within string
/usr/bin/msgfmt: found 1 fatal error
CMakeFiles/i18n-create-mo-pl_PL.dir/build.make:57: recipe for target 'i18n-create-mo-pl_PL' failed
````

smartavionics, 2018-10-02:
````
diff --git a/resources/i18n/pl_PL/fdmprinter.def.json.po b/resources/i18n/pl_PL/fdmprinter.def.json.po
index 53aa32009..fc7142598 100644
--- a/resources/i18n/pl_PL/fdmprinter.def.json.po
+++ b/resources/i18n/pl_PL/fdmprinter.def.json.po
@@ -6,7 +6,7 @@
 msgid ""
 msgstr ""
 "Project-Id-Version: Cura 3.5\n"
-"Report-Msgid-Bugs-To: r.dulek@ultimaker.com
+"Report-Msgid-Bugs-To: r.dulek@ultimaker.com\n"
 "POT-Creation-Date: 2018-09-19 17:07+0000\n"
 "PO-Revision-Date: 2018-09-21 21:52+0200\n"
 "Last-Translator: 'Jaguś' Paweł Jagusiak, Andrzej 'anraf1001' Rafalski and Jakub 'drzejkopf' Świeciński\n"

````

diegopradogesto, 2018-10-02:
Oh, yes, I fixed it this afternoon (here https://github.com/Ultimaker/Cura/commit/a99241d20bcfb6a46b920d3e0257083e04291323). Thanks for the heads up!

closed 2018-10-02 by diegopradogesto

smartavionics, 2018-10-03:
Still not OK in master branch at 0737 03-10-18.

smartavionics, 2018-10-03:
````
git diff upstream/3.5...upstream/master ../resources/i18n/pl_PL/fdmprinter.def.json.po
diff --git a/resources/i18n/pl_PL/fdmprinter.def.json.po b/resources/i18n/pl_PL/fdmprinter.def.json.po
index a8b07e032..53aa32009 100644
--- a/resources/i18n/pl_PL/fdmprinter.def.json.po
+++ b/resources/i18n/pl_PL/fdmprinter.def.json.po
@@ -6,7 +6,7 @@
 msgid ""
 msgstr ""
 "Project-Id-Version: Cura 3.5\n"
-"Report-Msgid-Bugs-To: r.dulek@ultimaker.com"
+"Report-Msgid-Bugs-To: r.dulek@ultimaker.com
 "POT-Creation-Date: 2018-09-19 17:07+0000\n"
 "PO-Revision-Date: 2018-09-21 21:52+0200\n"
 "Last-Translator: 'Jaguś' Paweł Jagusiak, Andrzej 'anraf1001' Rafalski and Jakub 'drzejkopf' Świeciński\n"
````

nallath, 2018-10-03:
er... No idea what the hell went wrong there, but thanks for spotting it. For some reason it was on 3.5, pulling 3.5 into master didn't do anything.