Ultimaker/Cura #4460: Unable to add custom materials, and materials are entirely empty

Zer0t3ch, 2018-10-02:
**Application Version**
3.4 (master)

**Platform**
Linux (OpenSUSE Tumbleweed)

**Printer**
Should be unrelated, but I use a custom AM8

**Steps to Reproduce**
1. Start Cura
2. Enter settings
3. In the "Materials" tab, click "Create"

**Actual Results**
Instant crash to desktop

**Expected results**
"Create Material" dialog

**Additional Information**
```
Terminal output:
2018-10-02 06:17:12,037 - DEBUG - [MainThread] cura.AutoSave._onTimeout [49]: Autosaving preferences, instances and profiles
2018-10-02 06:17:16,147 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]: An uncaught error has occurred!
2018-10-02 06:17:16,148 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: Traceback (most recent call last):
2018-10-02 06:17:16,148 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/usr/lib/python3.6/site-packages/cura/Machines/MaterialManager.py", line 549, in createMaterial
2018-10-02 06:17:16,148 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     self.duplicateMaterial(material_group.root_material_node,
2018-10-02 06:17:16,148 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: AttributeError: 'NoneType' object has no attribute 'root_material_node'
```

fieldOfView, 2018-10-02:
Did you check out the fdmmaterials repo?
https://github.com/ultimaker/fdm_materials
It goes in resources/materials

LipuFei, 2018-10-10:
Hi @Zer0t3ch , could you tell us how you created your printer in Cura? Did you create a "Custom FDM printer" and modify the settings from there, or via some other way?

commit 8f9d6bee (2018-11-26): Document dependency on fdm_materials

Without it, Cura simply won't start if you had a printer loaded with material profiles.

Fixes #4460.
closed 2018-11-26 by Ghostkeeper