Ultimaker/Cura #4465: Cura 3.5 doubled estimated print time

jimloase, 2018-10-03:
<Just tried to print an old item with 3.5 The estimated print time is doubled.

**Application Version**
3.50
**Platform**
64 bit windows 10

**Printer**
anycubic I3 Mega
[AI3M_Rotating Base Body7.zip](https://github.com/Ultimaker/Cura/files/2442576/AI3M_Rotating.Base.Body7.zip)

**Steps to Reproduce**
let it slice

**Actual Results**
time doubled from 3.4.1

**Expected results**
expect 1/2 the print time

**Additional Information**

Vandrasc, 2018-10-03:
Hi @jimloase, when compared with 3.4 my printing time decreased even with 1 minute. 
I did notice that the anycube I3 Mega has different profiles than in 3.4. The default profile in 3.4 has a 0.2 LH and in 3.5 the default profile is Fine which is printed with a 0.1 LH, that explains the double printing time. 
Could you confirm that you are using the same quality profiles for both cura versions? 
Thank you!

jimloase, 2018-10-03:
You are correct in that the layer height was changed to .1. That then 
begs the question why would the layer height be changed in a profile I 
had created and saved in 3.4. And what other changes need I search for 
when using a profile I created in 3.4 in 3.5.

Thank You

<mention@noreply.github.com>
Sent: 10/3/2018 1:17:02 PM
(#4465)

BagelOrb, 2018-10-04:
You mean that your own editen/created profile was changed by the update? Sounds like a bug!

Could you please attach a project file, or export the profile you were working on?

BagelOrb, 2018-10-04:
Put files in a zip and drag it in your new comment here on GitHub.