Ultimaker/Cura #4466: Cura 3.5 does not launch on OS X 10.9 Mavericks

probonopd, 2018-10-03:
**Application Version**
3.5

**Platform**
OS X 10.9

**Steps to Reproduce**
Try to launch Cura

**Actual Results**
Does not launch. Does not give an error either.

**Expected results**
Application launches. It worked in the previous version, 3.4.1.

**Additional Information**
Older Mac models may not be able to run later versions of the system software, so being able to run on older releases of OS X is important.

Vandrasc, 2018-10-03:
Hi @probonopd , silly question but did you tried with a clean install? You can remove your Cura user directory from Application Support and try launching 3.5 again, make sure to make a copy of your directory, just in case you don't want to lose your old configuration, qualities etc. 
Thank you!

probonopd, 2018-10-03:
Yes. Removing the cura directory from Application Support does not make 3.5 launch.

```
users-imac:~ user$ /Volumes/Ultimaker\ Cura-3.5.0-Darwin/Ultimaker\ Cura.app/Contents/MacOS/cura 
users-imac:~ user$ 
```

Just nothing happens, even when launched from the command line.

probonopd, 2018-10-04:
@LipuFei 3.5 beta has the same issue for me.

smartavionics, 2018-10-04:
Hi, does my experimental release that was built on 10.10.5 work for you?

https://www.dropbox.com/s/jpoqc1351ycfo57/cura-mb-master-20180926.dmg?dl=0

probonopd, 2018-10-04:
Unfortunately, no @smartavionics

smartavionics, 2018-10-04:
Oh, sorry to hear that. Same as the others, silently fails to launch?

probonopd, 2018-10-04:
Yes, exactly, even on the command line. Strange. 

But using `dtruss` I get something:

```
users-imac:~ user$ sudo dtruss /Volumes/cura-mb-master/cura-mb-master.app/Contents/MacOS/cura 2>&1
(...)
open("/Volumes/cura-mb-master/cura-mb-master.app/Contents/MacOS/imp.py\0", 0x1000000, 0x1B6)		 = -1 Err#2
open("/Volumes/cura-mb-master/cura-mb-master.app/Contents/MacOS/lib/python35.zip/imp.py\0", 0x1000000, 0x1B6)		 = -1 Err#20
open("/Volumes/cura-mb-master/cura-mb-master.app/Contents/MacOS/lib/python3.5/imp.py\0", 0x1000000, 0x1B6)		 = -1 Err#2
open("/Volumes/cura-mb-master/cura-mb-master.app/Contents/MacOS/lib/python3.5/plat-darwin/imp.py\0", 0x1000000, 0x1B6)		 = -1 Err#2
open("/Volumes/cura-mb-master/cura-mb-master.app/Contents/MacOS/lib/python3.5/lib-dynload/imp.py\0", 0x1000000, 0x1B6)		 = -1 Err#2
write(0x5, "Error in sys.excepthook:\nTraceback (most recent call last):\n  File \"<frozen importlib._bootstrap>\", line 969, in _find_and_load\n  File \"<frozen importlib._bootstrap>\", line 958, in _find_and_load_unlocked\n  File \"<frozen importlib._bootstrap>\", line 673, i", 0x925)		 = 2341 0
sigaction(0x2, 0x7FFF524CA5F8, 0x7FFF524CA620)		 = 0 0
```

smartavionics, 2018-10-04:
I'd love to be able to help but this stuff is completely alien to me. I am not either a Mac user or a Python programmer. Sorry.

probonopd, 2018-10-04:
Thanks for your help so far @smartavionics. 

By the way, I do realize that OS X 10.9 is quite old and not officially supported anymore - but before 3.5 ran Cura fine... and upgrading the OS would mean to throw away an otherwise well-running machine

LipuFei, 2018-10-05:
Hi @probonopd , could you try the following? I don't know for sure but it could be related to the `PostProcessingPlugin`:
 - Copy the ".app" to your local hard drive
 - In the terminal, remove "PostPorcessingPlugin" from that ".app", should be in `.app/Contents/Resources/plugins/plugins/PostProcessingPlugin`
 - Try start Cura again

probonopd, 2018-10-05:
Thanks @LipuFei but that does not change anything. In fact not even removing the whole `.app/Contents/Resources/plugins/plugins/` changes anything.

odpad, 2018-10-06:
the same issue on my OS X 10.9.5 :-(

Altairama, 2018-10-09:
Same problem here, it doesn´t launch in a mid 2010 macbook pro mid 2010 with mavericks 10.9.5

LipuFei, 2018-10-09:
Hi all, I'm not sure what's causing the problem. We've just made some changes to our build servers. Could you try a new 3.5 test build for us? Thanks in advance.

Download Link: https://we.tl/t-kH9Nxy0zsZ

LipuFei, 2018-10-10:
@probonopd @odpad @Altairama ↑

Altairama, 2018-10-10:
> Hi all, I'm not sure what's causing the problem. We've just made some changes to our build servers. Could you try a new 3.5 test build for us? Thanks in advance.
> 
> Download Link: https://we.tl/t-kH9Nxy0zsZ

Just checked it, sadly, still get the same issue

LipuFei, 2018-10-10:
@Altairama Is there anything in the log file?

probonopd, 2018-10-10:
Thanks @LipuFei, highly appreciate that you are looking into this. Using the version from https://we.tl/t-kH9Nxy0zsZ it does not launch on OS X 10.9.5 (13F34), and even when running from the Terminal, there is no error log.

Is there a way to make it more verbose?

When running `dtruss` I am getting a lot of `-1 Err#` errors, ending in

```
write(0x5, "Error in sys.excepthook:\nTraceback (most recent call last):\n  File \"<frozen importlib._bootstrap>\", line 969, in _find_and_load\n  File \"<frozen importlib._bootstrap>\", line 958, in _find_and_load_unlocked\n  File \"<frozen importlib._bootstrap>\", line 673, i", 0x90D)		 = 2317 0
sigaction(0x2, 0x7FFF5F2AD558, 0x7FFF5F2AD580)		 = 0 0
```

LipuFei, 2018-10-11:
Hi @probonopd , thanks for more info on this. I compared the binaries files in 3.4.1 and 3.5 and found some libraries are not compiled to target OS X 10.7 but 10.10 or even higher. I think that's the problem. I'm working on it now.

LipuFei, 2018-10-11:
```
Cura 3.4.1:
----
 > otool -l libQt5Core.5.dylib | grep -b5 LC_VERSION_MIN_MACOSX
8268-Load command 7
8283-     cmd LC_UUID
8300- cmdsize 24
8312-    uuid 25ED4661-4A7A-3006-96BB-6A58537B44C2
8358-Load command 8
8373:      cmd LC_VERSION_MIN_MACOSX
8405-  cmdsize 16
8418-  version 10.9
8433-      sdk 10.11
8449-Load command 9
8464-      cmd LC_SOURCE_VERSION
```

```
Cura 3.5
 > otool -l libQt5Core.5.dylib | grep -b5 LC_VERSION_MIN_MACOSX
8489-Load command 7
8504-     cmd LC_UUID
8521- cmdsize 24
8533-    uuid 43568DF7-2CEE-3414-A0BA-3CF236B5E5D3
8579-Load command 8
8594:      cmd LC_VERSION_MIN_MACOSX
8626-  cmdsize 16
8639-  version 10.10
8655-      sdk 10.11
8671-Load command 9
8686-      cmd LC_SOURCE_VERSION
```

LipuFei, 2018-10-11:
JIRA ticket: CURA-5812

LipuFei, 2018-10-11:
Hi all, just created a new 3.5.1 test build, trying to fix the compatibility problem with OSX 10.9. Could you test it for me see if it works on 10.9? Thanks a lot!

**Link:** https://we.tl/t-4bkuQIKu1O

LipuFei, 2018-10-11:
@odpad Could you click "Report" and copy-paste the information in the report here?

probonopd, 2018-10-11:
That `qt_apple_check_os_version()` looks suspicious...

LipuFei, 2018-10-12:
Hi @odpad @probonopd, could you share your log files? You can find them in
```
folder: /Users/<your-user-name>/Library/Application Support/cura
  - stdout.log
  - stderr.log
  - /3.5/cura.log
```

Altairama, 2018-10-12:
Same here, now it gives an error:

Process:         cura [2398]
Path:            /Applications/Ultimaker Cura.app/Contents/MacOS/cura
Identifier:      nl.ultimaker.cura
Version:         3.5 (3.5.1-test)
Code Type:       X86-64 (Native)
Parent Process:  launchd [138]
Responsible:     cura [2398]
User ID:         501

Date/Time:       2018-10-12 11:06:14.383 +0200
OS Version:      Mac OS X 10.9.5 (13F1911)
Report Version:  11
Anonymous UUID:  F82A7D29-8533-7CCA-22E8-67E377F080E2

Sleep/Wake UUID: 1B212EA2-903A-402A-A27A-2F5E0B1FE8FC

Crashed Thread:  0  Dispatch queue: com.apple.main-thread

Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff88f79292 __kill + 10
1   libsystem_platform.dylib      	0x00007fff908ca5aa _sigtramp + 26
2   libsystem_kernel.dylib        	0x00007fff88f79866 __pthread_kill + 10
3   libsystem_c.dylib             	0x00007fff8cccdb26 abort + 125
4   libQt5Core.5.dylib            	0x000000010e86f7ad qt_apple_check_os_version() + 317
5   libQt5Core.5.dylib            	0x000000010e7a8a49 QCoreApplicationPrivate::QCoreApplicationPrivate(int&, char**, unsigned int) + 153
6   libQt5Gui.5.dylib             	0x000000010eed3e4e QGuiApplicationPrivate::QGuiApplicationPrivate(int&, char**, int) + 14
7   libQt5Widgets.5.dylib         	0x000000010f9545b8 QApplication::QApplication(int&, char**, int) + 56
8   PyQt5.QtWidgets.so            	0x000000010f43419f init_type_QApplication(_sipSimpleWrapper*, _object*, _object*, _object**, _object**, _object**) + 175
9   sip.so                        	0x00000001067921fd sipSimpleWrapper_init + 183
10  nl.ultimaker.cura             	0x0000000105fd6c4c 0x105f6c000 + 437324
11  nl.ultimaker.cura             	0x0000000105f77ec5 PyObject_Call + 101
12  nl.ultimaker.cura             	0x000000010603eb60 PyEval_EvalFrameEx + 27536
13  nl.ultimaker.cura             	0x0000000106042bb2 0x105f6c000 + 879538
14  nl.ultimaker.cura             	0x0000000106037fae PyEval_EvalCodeEx + 78
15  nl.ultimaker.cura             	0x0000000105fa152d 0x105f6c000 + 218413
16  nl.ultimaker.cura             	0x0000000105f77ec5 PyObject_Call + 101
17  nl.ultimaker.cura             	0x000000010603eb60 PyEval_EvalFrameEx + 27536
18  nl.ultimaker.cura             	0x0000000106042bb2 0x105f6c000 + 879538
19  nl.ultimaker.cura             	0x0000000106037fae PyEval_EvalCodeEx + 78
20  nl.ultimaker.cura             	0x0000000105fa152d 0x105f6c000 + 218413
21  nl.ultimaker.cura             	0x0000000105f77ec5 PyObject_Call + 101
22  nl.ultimaker.cura             	0x0000000105f8dc0d 0x105f6c000 + 138253
23  nl.ultimaker.cura             	0x0000000105f77ec5 PyObject_Call + 101
24  nl.ultimaker.cura             	0x0000000105fd6bbe 0x105f6c000 + 437182
25  nl.ultimaker.cura             	0x0000000105fd3179 0x105f6c000 + 422265
26  nl.ultimaker.cura             	0x0000000105f77ec5 PyObject_Call + 101
27  nl.ultimaker.cura             	0x000000010603e066 PyEval_EvalFrameEx + 24726
28  nl.ultimaker.cura             	0x0000000106042bb2 0x105f6c000 + 879538
29  nl.ultimaker.cura             	0x0000000106037f51 PyEval_EvalCode + 81
30  nl.ultimaker.cura             	0x000000010605b176 PyImport_ExecCodeModuleObject + 390
31  nl.ultimaker.cura             	0x00000001060c229d 0x105f6c000 + 1401501
32  nl.ultimaker.cura             	0x0000000105fc0f95 PyCFunction_Call + 165
33  nl.ultimaker.cura             	0x000000010603e3c0 PyEval_EvalFrameEx + 25584
34  nl.ultimaker.cura             	0x00000001060433a2 0x105f6c000 + 881570
35  nl.ultimaker.cura             	0x000000010603dd0f PyEval_EvalFrameEx + 23871
36  nl.ultimaker.cura             	0x00000001060433a2 0x105f6c000 + 881570
37  nl.ultimaker.cura             	0x000000010603dd0f PyEval_EvalFrameEx + 23871
38  nl.ultimaker.cura             	0x00000001060433a2 0x105f6c000 + 881570
39  nl.ultimaker.cura             	0x000000010603dd0f PyEval_EvalFrameEx + 23871
40  nl.ultimaker.cura             	0x0000000106042bb2 0x105f6c000 + 879538
41  nl.ultimaker.cura             	0x0000000106037fae PyEval_EvalCodeEx + 78
42  nl.ultimaker.cura             	0x0000000105fa152d 0x105f6c000 + 218413
[... 352 lines cut ...]
__TEXT                            211.9M
__UNICODE                           544K
mapped file                        22.4M
shared memory                         4K
===========                      =======
TOTAL                             550.9M
TOTAL, minus reserved VM space    518.6M
 

Model: MacBookPro7,1, BootROM MBP71.0039.B0E, 2 processors, Intel Core 2 Duo, 2.4 GHz, 8 GB, SMC 1.62f7
Graphics: NVIDIA GeForce 320M, NVIDIA GeForce 320M, PCI, 256 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1067 MHz, 0x0198, 0x393955353432382D3034312E4130314C4620
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1067 MHz, 0x0198, 0x393955353432382D3034312E4130314C4620
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x8D), Broadcom BCM43xx 1.0 (5.106.98.100.22)
Bluetooth: Version 5.2.7f4 15858, 3 services, 15 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Samsung SSD 840 EVO 250GB, 250,06 GB
Serial ATA Device: HL-DT-ST DVDRW  GS23N
USB Device: Built-in iSight
USB Device: Internal Memory Card Reader
USB Device: BRCM2046 Hub
USB Device: Bluetooth USB Host Controller
USB Device: IR Receiver
USB Device: Apple Internal Keyboard / Trackpad
Thunderbolt Bus:

Altairama, 2018-10-12:
Also, in:
folder: /Users/<your-user-name>/Library/Application Support/cura
I can only find:
[image]
(I have instaled the last build 3.5.1)

LipuFei, 2018-10-15:
If you run "Console", switch to "cura", and then run Cura, does it show anything?
[image]
[image]

LipuFei, 2018-10-15:
Hi all, here's another build from today. Could you try it out? Thank you!

https://we.tl/t-8sVDV2s3pT

odpad, 2018-10-15:
still the same behavior - Ultimaker Cura 3.5.1 quit unexpectedly. ...

"stdout.log" is empty

this is cura's "stderr.log":

---

Fatal Python error: Aborted

Current thread 0x00007fff77a97310 (most recent call first):
  File "/Users/ultimaker/build/3.5-test-fix-mac-10.9/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 92 in __init__
  File "/Users/ultimaker/build/3.5-test-fix-mac-10.9/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 156 in __init__
  File "/Users/ultimaker/build/3.5-test-fix-mac-10.9/build/inst/bin/cura_app.py", line 137 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/3.5-test-fix-mac-10.9/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/3.5-test-fix-mac-10.9/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

---

and this is console log:

---

Process:         cura [17153]
Path:            /Applications/Ultimaker Cura.app/Contents/MacOS/cura
Identifier:      nl.ultimaker.cura
Version:         3.5 (3.5.1-test10.9)
Code Type:       X86-64 (Native)
Parent Process:  launchd [187]
Responsible:     cura [17153]
User ID:         501

Date/Time:       2018-10-15 17:01:27.659 +0200
OS Version:      Mac OS X 10.9.5 (13F1911)
Report Version:  11
Anonymous UUID:  1348D877-5E02-29C8-4103-0218C9E3E16E

Sleep/Wake UUID: 1E9A7523-3C5D-498B-AF5A-E8EFD6D8D7B5

Crashed Thread:  0  Dispatch queue: com.apple.main-thread

Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff89859292 __kill + 10
1   libsystem_platform.dylib      	0x00007fff88ff45aa _sigtramp + 26
2   libsystem_kernel.dylib        	0x00007fff89859866 __pthread_kill + 10
3   libsystem_c.dylib             	0x00007fff87b6ab26 abort + 125
4   libQt5Core.5.dylib            	0x0000000119b7a7ad qt_apple_check_os_version() + 317
5   libQt5Core.5.dylib            	0x0000000119ab3a49 QCoreApplicationPrivate::QCoreApplicationPrivate(int&, char**, unsigned int) + 153
6   libQt5Gui.5.dylib             	0x000000011b31fe4e QGuiApplicationPrivate::QGuiApplicationPrivate(int&, char**, int) + 14
7   libQt5Widgets.5.dylib         	0x000000011bd955b8 QApplication::QApplication(int&, char**, int) + 56
8   PyQt5.QtWidgets.so            	0x000000011b87519f init_type_QApplication(_sipSimpleWrapper*, _object*, _object*, _object**, _object**, _object**) + 175
9   sip.so                        	0x000000010d98e1fd sipSimpleWrapper_init + 183
[... 387 lines cut ...]
Writable regions: Total=220.7M written=15.4M(7%) resident=35.2M(16%) swapped_out=0K(0%) unallocated=185.5M(84%)
 
REGION TYPE                      VIRTUAL
===========                      =======
Dispatch continuations             16.0M
Kernel Alloc Once                     4K
MALLOC                             95.1M
MALLOC (admin)                       32K
MALLOC_LARGE (reserved)             384K        reserved VM address space (unallocated)
STACK GUARD                          24K
Stack                              66.6M
VM_ALLOCATE                          36K
VM_ALLOCATE (reserved)             96.0M        reserved VM address space (unallocated)
__DATA                             25.1M
__IMAGE                             528K
__LINKEDIT                         86.3M
__TEXT                            211.9M
__UNICODE                           544K
mapped file                        22.4M
shared memory                         4K
===========                      =======
TOTAL                             620.9M
TOTAL, minus reserved VM space    524.5M

---

LipuFei, 2018-10-16:
Hi all, thanks for your helps on testing this. Unfortunately, I have some bad news... We switched to use Qt 5.10.1 in Cura 3.5, and Qt 5.10.1 only supports Mac OS X 10.10+ ([link](https://doc.qt.io/qt-5/supported-platforms-and-configurations.html) They say 5.10.1 supports 10.11+ but it seems to work as well on 10.10). This is not as easy as to set the minimum support macosx version to 10.9 and be done with it. In Qt 5.10.1, many new OS X functions which are only available in 10.10+ are utilized, and this makes it incompatible with OS X 10.9 and earlier.

We will discuss this problem. So far, the workaround is to use 3.4.1 if you have a older OS X such as 10.9 and before.

probonopd, 2018-10-16:
> We switched to use Qt 5.10.1 in Cura 3.5, and Qt 5.10.1 only supports Mac OS X 10.10+ 

The question is, does Cura really benefit so much from switching to Qt 5.10.1.

> We will discuss this problem. So far, the workaround is to use 3.4.1 if you have a older OS X such as 10.9 and before.

Thank you for your willingness to consider this problem. If this was Windows or Linux the answer would be easy, "just update your operating system", but given that Apple has a policy of preventing users from installing the latest OSes on (otherwise perfectly capable) machines after a couple of years, there are users out there for which this makes the difference between having a perfectly capable computer and sitting on a pile of electronic waste.

LipuFei, 2018-10-18:
Hi all, we managed to create a Cura build with Qt 5.9 that seems to work on the later OSX versions (>=10.10). It's not using the 3.5 code base but the current in-development code base. Could you help us to test it? You can find the link below. Many thanks!

**Link:** https://we.tl/t-wirdd8iONj

odpad, 2018-10-27:
any progress on this issue?

Vandrasc, 2018-11-01:
Hi all, a fix was created to handle OSX version 10.10, but due to the QT upgrade, unfortunately, we are dependent on QT, users with older versions won’t be able to run Ultimaker Cura on their system without a system update. This applies to OSX version 10.09 and 10.08. 

Can any of you confirm the fix for 10.10 version?

@LipuFei , is there anything else to be added?

LipuFei, 2018-11-01:
Hi all, as @Vandrasc said, unfortunately we now depend on Qt 5.10 and cannot go back to 5.8. This means that we can no longer support OSX versions earlier than 10.10.

odpad, 2018-11-01:
:-(

closed 2018-11-01 by ChrisTerBeke