Ultimaker/Cura #4467: New Ultimaker Cura Settings Guide plugin breaks in dark theme

renssies, 2018-10-03:
The new Cura Settings Guide plugin made by Multimaker doesn't work when the dark theme is enabled. The setting names on the right are unreadble and the text is hard to read because of the dark background. See the following screenshots:

**Ultimaker theme (normal/default)**
[image]

**Ultimaker dark theme**
[image]

**Application Version**
3.5

**Platform**
macOS 10.14 (Mojave)

**Printer**
None

**Steps to Reproduce**
1. Install "Cura Settings Guide" from the toolbox
2. Enable Cura dark theme in Preferences
3. Restart Cura to apply the settings guide and dark theme
4. Open the settings guide from "Extensions --> Cura Settings Guide"
5. Observe the guide being unreadable.

**Actual Results**
The setting names on the right of the Cura settings guide are unreadable.
The text is hard to read due to black text on a dark background.  

**Expected results**
The setting names should be readable when in dark or light theme.
The text should have a white background or the text should have a white/light color in Dark theme

ChrisTerBeke, 2018-10-03:
ping @alekseisasin

ChrisTerBeke, 2018-10-03:
Fyi, the settings guide plugin source code can be found at https://github.com/alekseisasin/CuraSettingsGuidePlugin.

alekseisasin, 2018-10-03:
@renssies , Thank you for reporting this issue. 
Btw, the source code was moved to this repository: https://github.com/alekseisasin/SettingsGuide

alekseisasin, 2018-10-04:
It is fine that you posted here. I think later it will be moved to Ultimkaer GitHub.

alekseisasin, 2018-10-29:
The Settings Guide plugin keeps the same layout colours for different Cura Themes. The new version is 1.0.1. You already can download it.

closed 2018-10-29 by alekseisasin