Ultimaker/Cura #4468: Ender 3 Profile not correct 3.5

tauceti82, 2018-10-03:
Hi,

the new Ender 3 Profile has a wrong fine profile.
It has the same layer height as normal 1.5 whereas CR10 have 1mm
Fine also has a lower wall thickness as normal profile.
Please recheck. I haven'T compared all values. 
It would be nice to have the same CR10 fine profile in the Ender 3 printer too

ChrisTerBeke, 2018-10-03:
It's up to community/Creality to provide the correct values for 3rd party machines.

stelgenhof, 2018-10-04:
This is because Cura 3.5 includes an incorrect version of the Ender-3 machine definition. The incorrect version was corrected and committed but apparently didn't make it in the 3.5 release.

You can see in the master branch (https://github.com/Ultimaker/Cura/tree/master/resources) that the correct profile and extruder train definition are present.

closed 2018-10-04 by ChrisTerBeke