Ultimaker/Cura #4469: CURA 3.5.0    Temperatur Fehler im GCode

lwbsbde, 2018-10-03:
Ich habe im Start GCode folgenden Eintrag
 "M104 S{material_print_temperature}"

Aber egal was ich in Cura als Drucktemperatur einstelle ( zum Beispiel 190 oder 240 ),
Im erzeugten GCode steht immer " M104 S210 ".

Bei Cura 3.4.  wurde  die Temperatur korrekt übernommen.

ChrisTerBeke, 2018-10-03:
Please post your question/issue in English so all the developers can read it.

ChrisTerBeke, 2018-10-03:
Does `M104 S{material_print_temperature, 0}` work for you?

lwbsbde, 2018-10-03:
Thank you , it works  :-)

ChrisTerBeke, 2018-10-03:
Thanks for testing. It does seem we have a regression then because 0 should be the default... @Vandrasc

LazaroFilm, 2018-10-04:
Had the same issue here. Yes, adding the `, 0` solved it, but why did this change behavior in the first place?

fieldOfView, 2018-10-04:
@LazaroFilm, do you also have a multi-extrusion printer?

What is the adhesion extruder set to?

ChrisTerBeke, 2018-10-04:
@LazaroFilm it shouldn't have, that's why I mentioned a regression occured.

Ketchu13, 2018-10-04:
> Thanks for testing. It does seem we have a regression then because 0 should be the default... @Vandrasc

Hi,
my printer profile was with 2 extruders, reduced to only one (I don't have anymore dual extrusion installed on the printer)
using `{material_print_temperature, 0}` return the default not the first layer temperature value
using `{material_print_temperature_layer_0, 0}`  return the correct value

Appesteijn, 2018-10-04:
CURA-5793

diegopradogesto, 2018-10-05:
@lwbsbde are you using a custom printer definition? can you provide a project file or your configuration folder to deeply analyse what is going on here?

Thanks.

lwbsbde, 2018-10-06:
yes, I using a custom Printer definition. what folder do you mean?

lekoder, 2018-10-07:
This was a nasty surprise, with temperature defaulting to 210'C this caused my printer to grind trough the filament.

gwax, 2018-10-07:
I can confirm this issue as well. I can also confirm that the suggested workaround of specifying extruder 0 explicitly in the start gcode (`{material_print_temperature_layer_0, 0}`) worked for me.

nallath, 2018-10-08:
Fixed in https://github.com/Ultimaker/Cura/pull/4520

closed 2018-10-08 by nallath

ctmakro, 2018-10-08:
so I just ran into the same problem(downloaded 3.5 installer); happy to see it solved an hour ago.