Ultimaker/Cura #4470: CURA 3.5.0 Anpassschichten

lwbsbde, 2018-10-03:
Die Anpassschichten werden  zu spät geändert. 

Bei Überhängen/Schrägen wird die erste Schicht noch mit maximaler Dicke gedruckt.

Die Anpassung der Schichtdicke müsste 2 Schichten eher beginnen.

ChrisTerBeke, 2018-10-03:
Please post your question/issue in English so all the developers can read it.

ChrisTerBeke, 2018-10-04:
This is caused by the engine not looking very far ahead when calculating the layer heights (which is done before almost all other calculations). Changing this behavior would be quite a challenge.

Ghostkeeper, 2021-11-30:
This is being tracked through https://github.com/Ultimaker/Cura/issues/6641. Even though this feature request was earlier, I'll close it in favour of the other which has more information now. We're also tracking this in Ultimaker's internal Jira tracker as CURA-7026 (but only Ultimaker employees can see that).

closed 2021-11-30 by Ghostkeeper