Ultimaker/Cura #4474: [3.5]machine extruder chain not specified in Creality ender 3 definition file

ilporcupine, 2018-10-04:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

SaskiFX, 2018-10-04:
Looks like the extruder is listed in the extruders folder, just didn't get mentioned in the definition for the printer.

stelgenhof, 2018-10-04:
This is because Cura 3.5 includes an incorrect version of the Ender-3 machine definition. The incorrect version was corrected and committed but apparently didn't make it in the 3.5 release.

You can see in the master branch (https://github.com/Ultimaker/Cura/tree/master/resources) that the correct profile and extruder train definition are present.

BagelOrb, 2018-10-04:
So this is fixed already for the next release - if I understand correctly.

I can reopen if need be.

closed 2018-10-04 by BagelOrb

stelgenhof, 2018-10-04:
@BagelOrb Thanks for the quick response!