Ultimaker/Cura #4475: Can't load .SLDASM files anymore in Cura 3.5.0

gunthix, 2018-10-04:
**Application Version**
3.5.0

**Platform**
Windows 7

**Printer**
UM3

**Steps to Reproduce**
"Open file" --> try to open a .SLDASM file

**Actual Results**
Box says "Invalid File"

**Expected results**
Loading the .SLDASM file like normal in the previous Cura version

**Additional Information**
Love Cura, keep up the good work!

ChrisTerBeke, 2018-10-04:
SLDASM is not a natively supported file format, so it must have been supported via a plugin. Cura 3.5 introduced breaking changes for plugins so they have to be updated as well (most of them were during the Beta period). Please contact the author of the plugin that you're using for this format.

Appesteijn, 2018-10-04:
To open these files you need the Solidworks plugin (if I'm correct). When we get an updated Solidworks plugin that works (there are breaking plugin changes in Cura 3.5) we will get it in the ToolBox asap.

closed 2018-10-04 by ChrisTerBeke