Ultimaker/Cura #4476: Nobody responds to cura-build-environment issues

smartavionics, 2018-10-04:
**Application Version**

master

**Platform**

Any

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

I have two open cura-build-environment issues and nobody has responded in any way. I really would appreciate some help if that is possible?

ChrisTerBeke, 2018-10-04:
I actually proposed closing those issue threads and converge everything here (so it get's actually look at), but others didn't deem it necessary, apparently now resulting in things getting overlooked...

My suggestion to you is to open all issues here and have the devs deal with it.

smartavionics, 2018-10-04:
Understood.

I'll close this now.

closed 2018-10-04 by smartavionics