Ultimaker/Cura #4477: Cura not responding

upatamby, 2018-10-04:
**Application Version**
Cura 3.5.0

**Platform**
Windows 10

**Printer**
Ultimaker 3 Extended

**Steps to Reproduce**
What happened:
I opened Cura and loaded 4 models. They were assigned to material A. I assigned one of the models to material B and merged this model with one of the others. I sent the printjob over the network to the printer. 
Then I opened the Monitor tab. After a while (15 minutes?) I clicked on the Prepare tab. Then the title bar showed "(not responding)".
Taskmanger showed about 30% processor usage and a high increasing memory usage. I waited until 5GB memory was used and then aborted the task.
Unfortunately I was not able to reproduce the issue.

**Actual Results**
Not responding after a while.

**Expected results**
A well functioning application.

**Additional Information**
It was the first time I used this new Cura version, right after I installed it.

alekseisasin, 2018-10-04:
We need to investigate this issue. We already had an issue related to enormous memory usage in the Monitor tab. If you will able to reproduce this issue can you share the log file?

alekseisasin, 2018-10-16:
DEV: CURA-5821

Ghostkeeper, 2018-12-21:
This was fixed in Cura 3.6.

closed 2018-12-21 by Ghostkeeper