Ultimaker/Cura #4479: Cura won't detect my Printer!

MeSheepMan, 2018-10-04:
**Application Version**
3.2.1

**Platform**
Win 10

**Printer**
Monoprice Delta mini

**Steps to Reproduce**
I open up Cura

**Actual Results**
It Detected my printer twice at two different occasions and won't find it again

**Expected results**
It's suppose to see that my printer's connected

**Additional Information**
[image]

MeSheepMan, 2018-10-04:
If anyone knows away to fix this that would be amazing!

ChrisTerBeke, 2018-10-04:
A fix for this was made a while back (https://github.com/Ultimaker/Cura/pull/3855), please try a newer version of Cura. As we don't have this printer we can't test it, so we have to go by the community contributions when it comes to 3rd party printers.

closed 2018-10-04 by ChrisTerBeke

MeSheepMan, 2018-10-05:
Thank you very much! (sorry I posted such a easy problem) It's now running and great at that. :D