Ultimaker/Cura #4481: save is gcode

smitty101, 2018-10-05:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

smitty101, 2018-10-05:
why wont the new cura let me save files as gcode

Vandrasc, 2018-10-05:
Hi @smitty101 , could you please fill in the above template so we can help you better? What printer are you using? Are there any errors in the log? Which platform are you using? 

Thank you!

ChrisTerBeke, 2018-10-05:
Did you try file -> export -> Gcode?

ChrisTerBeke, 2018-10-05:
Or after slicing, use 'save to file' in the bottom right?

Ghostkeeper, 2018-11-29:
We'll assume that the problem arose from the File menu being restyled. Please use "Export..." or just press the blue button in the bottom right hand corner.

closed 2018-11-29 by Ghostkeeper