Ultimaker/Cura #4483: [3.5] Messy infil lines

Pajacyk0v, 2018-10-05:
**Application Version**
3.5

**Platform**
Windows 10 x64

**Printer**
Custom - Reprap

**Steps to Reproduce**
1. select infill lines multiplier > 1
2. select connect infill lines

**Actual Results**
There is complete mess in preview mode. Infill lines are conected thru model.

**Expected results**
Connected multiple infill lines :]

**Additional Information**
[image]

rburema, 2018-10-05:
Hi @Pajacyk0v,
Could you attach the model please? I also see you're using a custom printer. Have you written a custom definition? It may help if you attach that as well. 
(Also, this is just a hunch, but what nozzle-diameter do you use?)
cheers, Remco.

Pajacyk0v, 2018-10-12:
> 
> 
> Hi @Pajacyk0v,
> Could you attach the model please? I also see you're using a custom printer. Have you written a custom definition? It may help if you attach that as well.
> (Also, this is just a hunch, but what nozzle-diameter do you use?)
> cheers, Remco.

Sorry for a delay. I had holidays :]
I use 0.4 nozzle. Files in attachement
[cura_3.5_infill_issue.zip](https://github.com/Ultimaker/Cura/files/2474319/cura_3.5_infill_issue.zip)

smartavionics, 2018-10-12:
Hi @Pajacyk0v , is it possible you could save the project (.3mf) file, zip that and attach to this issue? Thanks.

Pajacyk0v, 2018-10-13:
> 
> 
> Hi @Pajacyk0v , is it possible you could save the project (.3mf) file, zip that and attach to this issue? Thanks.

In attachment. Regards
[dobijak_simmeringu.zip](https://github.com/Ultimaker/Cura/files/2475602/dobijak_simmeringu.zip)

smartavionics, 2018-10-13:
Thanks for the project. I can confirm that the problem is fixed using my Cura which is based on the master branch so I would think that it will be fixed in a future release.

closed 2018-10-16 by alekseisasin

Pajacyk0v, 2018-10-16:
With which release? 3.5.1 doesn't fix this problem.
[image]

smartavionics, 2018-10-16:
The next major release, I would expect.

rburema, 2018-10-17:
@smartavionics is right. Some background info: Point-releases like 3.5.1 are only supposed to fix major sources of crashes and other complete showstoppers like that (anything else is a bonus), and more importantly, are on the same branch as the 3.5.0 release that was split off from master a while ago (one or more weeks before that release). The master-branch will be used for the next release.

diegopradogesto, 2018-10-31:
@Pajacyk0v the version 3.6 BETA is live from today and this issue should be fixed there. You can try the version and test you projects. If you still see something wrong, please let us know.