Ultimaker/Cura #4489: Unable to switch between separate  printer profiles

hitchabout, 2018-10-05:
I am running Cura 3.5 the latest download on a Windows 10 64bit. When i try to switch printers from anet a6 to ender 3; Cura 3.5 crashes.  i still have Cura 3.4.1 installed which i have to use. plus i use octoprint via wifi.

fieldOfView, 2018-10-05:
For crashes and similar issues, please attach the following information:

    (On Windows) The log as produced by dxdiag (start -> run -> dxdiag -> save output)
    The Cura GUI log file, located at
        %APPDATA%\cura\<Cura version>\cura.log (Windows), or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log
        $USER/Library/Application Support/cura/<Cura version>/cura.log (OSX)
        $USER/.local/share/cura/<Cura version>/cura.log (Ubuntu/Linux)

If the Cura user interface still starts, you can also reach this directory from the application menu in Help -> Show settings folder

hitchabout, 2018-10-05:
is this what's needed?
[cura.log](https://github.com/Ultimaker/Cura/files/2450254/cura.log)

fieldOfView, 2018-10-05:
Yes, thanks.

This seems to be the interesting bit:
```
2018-10-05 05:30:19,231 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]: An uncaught error has occurred!
2018-10-05 05:30:19,233 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: Traceback (most recent call last):
2018-10-05 05:30:19,235 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "X:\3.5\build\inst\lib\python3.5\site-packages\cura\Settings\SettingInheritanceManager.py", line 70, in getOverridesForExtruder
2018-10-05 05:30:19,237 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "X:\3.5\build\inst\lib\python3.5\site-packages\cura\Settings\SettingInheritanceManager.py", line 193, in _settingIsOverwritingInheritance
2018-10-05 05:30:19,239 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: AttributeError: 'NoneType' object has no attribute 'getAllKeys'
```

Also of note, the above critical error is preceeded by about 50 warnings like this:
```
2018-10-05 05:30:17,666 - WARNING - [MainThread] cura.Settings.SettingInheritanceManager.getOverridesForExtruder [61]: Unable to find extruder for current machine with index 0
```

hitchabout, 2018-10-05:
I'm a noob at this; however, willing to learn. i tried reinstalling it thinking it was a corrupted downloaded file.  so, it is a bug?

fieldOfView, 2018-10-05:
Probably, yes.

stelgenhof, 2018-10-07:
Unfortunately an newer version of the Ender-3 printer definition I submitted (with the correct extruder definition file) wasn't included in the 3.5 release. That is most likely the reason for this error.

hitchabout, 2018-10-07:
ok

hitchabout, 2018-10-07:
i was using the old cura FFF custom printer profile that i made up a few updates back. i just found Ender 3 on the list of preset printers under "other". that may have been the issue. now it's loaded and working.

closed 2018-10-07 by hitchabout