Ultimaker/Cura #4491: cura 3.5.0 fails to slice

PierreAuge, 2018-10-05:
**Application Version**
3.5.0

**Platform**
windows 10

**Printer**
custom fdm

**Steps to Reproduce**
attempt to slice more than one model at a time using tree supports

**Actual Results**
hangs indefinitely

**Expected results**
slicing

same platter and settings in 3.4.1 works with no issues

**Additional Information**

[image]

[CFFFP_X Roller.zip](https://github.com/Ultimaker/Cura/files/2450317/CFFFP_X.Roller.zip)

[cura.log](https://github.com/Ultimaker/Cura/files/2450330/cura.log)

alekseisasin, 2018-10-08:
DEV: CURA-5798

diegopradogesto, 2018-10-09:
Weird... it does slice to me with 3.5 in Windows 10
[image]

LipuFei, 2018-10-09:
Hi @PierreAuge , looks like you are running more than 1 Curas side by side. Could you try just running 3.5 see if the problem is there?

LipuFei, 2018-10-09:
Hi @PierreAuge , we have found that this is a bug in TreeSupport. You don't need to try running one Cura.

PierreAuge, 2018-10-09:
Thank you!                                                                                                                                                                                                                                                                                                                                        Sent from my BlackBerry 10 smartphone on the Bell network.                                                                                                                                                                                                                From: Lipu FeiSent: Tuesday, October 9, 2018 8:43 AMTo: Ultimaker/CuraReply To: Ultimaker/CuraCc: PierreAuge; MentionSubject: Re: [Ultimaker/Cura] cura 3.5.0 fails to slice (#4491)Hi @PierreAuge , we have found that this is a bug in TreeSupport. You don't need to try running one Cura.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Ultimaker/Cura/issues/4491#issuecomment-428176920",
"url": "https://github.com/Ultimaker/Cura/issues/4491#issuecomment-428176920",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [Ultimaker/Cura] cura 3.5.0 fails to slice (#4491)",
"sections": [
{
"text": "",
"activityTitle": "**Lipu Fei**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@LipuFei",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"Ultimaker/Cura\",\n\"issueId\": 4491,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"Ultimaker/Cura\",\n\"issueId\": 4491\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/Ultimaker/Cura/issues/4491#issuecomment-428176920"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 388543562\n}"
}
],
"themeColor": "26292E"
}
]

LipuFei, 2018-10-10:
Hi @PierreAuge , after discussed with the team, we decided to fix this for the next major release but not in `3.5.x`. As a workaround, could you try using the regular support instead of tree support? Tree support is still experimental.

PierreAuge, 2018-10-10:
Yes, already on it. Cheers,Pierre                                                                                                                                                                                                                                                                                                                                        Sent from my BlackBerry 10 smartphone on the Bell network.                                                                                                                                                                                                                From: Lipu FeiSent: Wednesday, October 10, 2018 4:32 AMTo: Ultimaker/CuraReply To: Ultimaker/CuraCc: PierreAuge; MentionSubject: Re: [Ultimaker/Cura] [3.5] CuraEngine can crash when Tree Support is enabled (#4491)Hi @PierreAuge , after discussed with the team, we decided to fix this for the next major release but not in 3.5.x. As a workaround, could you try using the regular support instead of tree support? Tree support is still experimental.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Ultimaker/Cura/issues/4491#issuecomment-428485464",
"url": "https://github.com/Ultimaker/Cura/issues/4491#issuecomment-428485464",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [Ultimaker/Cura] [3.5] CuraEngine can crash when Tree Support is enabled (#4491)",
"sections": [
{
"text": "",
"activityTitle": "**Lipu Fei**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@LipuFei",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"Ultimaker/Cura\",\n\"issueId\": 4491,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"Ultimaker/Cura\",\n\"issueId\": 4491\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/Ultimaker/Cura/issues/4491#issuecomment-428485464"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 388543562\n}"
}
],
"themeColor": "26292E"
}
]

closed 2018-10-19 by Ghostkeeper