Ultimaker/Cura #4495: 3.5.0 does not load file from USB Stick

Michaels-3dPrinting, 2018-10-05:
**Application Version**

Cura 3.5.0

**Platform**

**Printer**

Anet A3
**Steps to Reproduce**

Double click file in explorer and wait for cura to load 
**Actual Results**

gets to the basic cura screen then starts or rather does not load file and opens cura error window - Says - Cura.exe has stopped working - A problem caused the program to stop working correctly.
Windows will close the program and notify you if a solution is available.
Windows opens a window and says - Cura.exe - Application Error
The instruction at 0x5a4c446a referenced memory at 0x00000008. The memory could not be read.

**Expected results**

File to be loaded and show *.stl on workplane
**Additional Information**

Michaels-3dPrinting, 2018-10-08:
> Hi @Michaels-3dPrinting <https://github.com/Michaels-3dPrinting> , it works on our machine. Did you try different USB sticks? Can you reproduce the same issue with different models?
> Also, could you please share the log file.

Hi

 

Tell me what to try and I will have a go.

 

Have at the moment gone back to 3.4.1

 

As it works every time to me and if I have to worry about which usb stick it would be too much of problem as - is it an absolute or would certain conditions cause the usb stick to fail later?

 

Works fine with 3.4.1 no problems of usb files

 

Let me know what you wish me to do

 

Using windows 8.something

 

Regards

 

Michael

alekseisasin, 2018-10-09:
@Michaels-3dPrinting , do not worry about this, we will try to fix it as soon as possible. After an update of the core framework, some functionalities do not work properly on all Operation systems.

LipuFei, 2018-10-09:
For dev, JIRA ticket: CURA-5801

LipuFei, 2018-10-09:
Fixed in branch 3.5

closed 2018-10-09 by LipuFei