Ultimaker/Cura #4496: Cura 3.5 problems when using Save to file on MacOS High Sierra

nospam2k, 2018-10-05:
First save of Cura after opening, when choosing a directory, does not remember the previous directory.

Reproduce: Open Cura. Open an stl file. Slice. Click save to file. Change to Desktop. Save. Close Cura. Open an stl file. Slice. Click save to file. Directory will not be Desktop.

ChrisTerBeke, 2018-10-06:
That sounds more than an OS issue than a Cura issue (as Cura just calls the file save dialog of the OS).

nospam2k, 2018-10-07:
The previous versions didn't do this. It's not an OS issue as other programs don't do this. When you open the save dialog it always (at least for me) default to Macintosh HD and not the directory I previously used. Most programs save the last location used.

nospam2k, 2018-10-08:
I can provide a screen capture if you like. Be sure that you closed Cura in between saves. When I open Cura, it always defaults to Macintosh HD regards of what was previously chosen. Also, be sure that you are using MacOS High Sierra.

LipuFei, 2018-10-09:
For dev, JIRA ticket: CURA-5803

LipuFei, 2018-10-10:
Hi @nospam2k , it seems to be a bug in Qt (or Mac OS X), which is the GUI framework we use to build Cura. We could not find a workaround so far and decided put this on "Deferred". Appreciate it a lot if someone can work on a solution for this.

closed 2018-10-10 by LipuFei