Ultimaker/Cura #4497: Cura 3.5 will not import Prusa i3 Mk3 materials profile

handyandyS3, 2018-10-05:
Cura 3.5 loads the Prusa i3 Mk3 profiles for the Printer and image fine, and also the layer heights profile. It says it has loaded the materials profile but none of the three Prusa materials is displayed when you look at the material choices.

ChrisTerBeke, 2018-10-06:
Did you select the Prusa i3 as active printer? Only then will you see the material appear in the list (it's filtered on compatibility with active machine).

handyandyS3, 2018-10-07:
Thanks for reply. Yes I did that. Also just uninstalled everything and re-installed it... just the same. It says it has imported the materials correctly but will then not display them at all in the materials manager????

⁣Sent from BlueMail ​

LipuFei, 2018-10-08:
Hi @handyandyS3 , I cannot open the prusa3d link from @talldonkey . As explained by @talldonkey , we have changed how the profiles work in Cura, so it could well be the case that the provided 3rd party profiles don't work for 3.5. Could you check if that's the case?

talldonkey, 2018-10-08:
I fixed the link (https://manual.prusa3d.com/Guide/How+to+import+profiles+to+Cura+(Windows+&+macOS)/703)

LipuFei, 2018-10-09:
Hi @talldonkey , thanks for the link. I've checked the files for MK2S and MK3, indeed they are no longer correct in Cura 3.5. I notified the page creator(s) about the breaking changes so they can update the page and the profiles to make them work in Cura 3.5.

LipuFei, 2018-10-09:
@talldonkey @handyandyS3 I'll close this ticket now. Sorry that you have to wait for prusa to update the printer profiles for Cura 3.5. At the meantime, you can use 3.4.1.

closed 2018-10-09 by LipuFei

handyandyS3, 2018-10-09:
Hi, Would it not be possible for me to edit the file into 3.5 format?

⁣Sent from BlueMail ​

LipuFei, 2018-10-09:
@handyandyS3 I haven't gone through all files thoroughly, so below is a fix for the most obvious problem I have found:

The printers are missing extruder definition files for Cura 3.5. Take the profiles for MK2S as an exmpale, you can find the following files in `MK2S_MK25_cura_profiles/01_Cura_printer_profiles`:
 - `Original_Prusa_i3_MK2S_MK25.def.json`
This is a JSON file that defines the machine, but there also needs to be one for its extruder(s).

For details, you can take a look at the following files:
 - https://github.com/Ultimaker/Cura/blob/master/resources/definitions/prusa_i3_mk2.def.json
 - https://github.com/Ultimaker/Cura/blob/master/resources/extruders/prusa_i3_mk2_extruder_0.def.json

In the case of MK2S, you can copy-and-paste https://github.com/Ultimaker/Cura/blob/master/resources/extruders/prusa_i3_mk2_extruder_0.def.json to create an extruder definition file for `Original_Prusa_i3_MK2S_MK25.def.json`.

(1) Modify `Original_Prusa_i3_MK2S_MK25.def.json` to include a field like this:
```
# YOU NEED TO CHANGE THIS. I just copied it from prusa_i3_mk2.def.json
"machine_extruder_trains":
{
    "0": "prusa_i3_mk2_extruder_0"
}
```
(2) Modify the extruder file you just created to contain the correct names and machine reference `Original_Prusa_i3_MK2S_MK25`.

For installation, I'll use Windows as an example. For the printer JSON file `Original_Prusa_i3_MK2S_MK25.def.json`, it's should be copied into `resources/definitions`. For the extruder JSON file that's just created, it needs to be in `resources/extruders`.

Hope this helps!