Ultimaker/Cura #4499: [3.4.1/3.5] Profile corruption warning when deleting any printer profile

Liger0, 2018-10-05:
**Application Version**
3.4.1 and 3.5, I didn't try others

**Platform**
Windows 10 x64 and Ubuntu x64

**Printer**
Cr-10

**Steps to Reproduce**
Add a new printer profile, switch it to the active one. Delete the old one. Now you'll read the error referring to a filament brand you have.

**Actual Results**
Nothing happens.

**Expected results**
An error suggests to reset everything.

**Additional Information**
[image]

alekseisasin, 2018-10-08:
@Liger0 ,
I have tried to reproduce the reported issue, my steps were:
1. Started Cura, at first start, I added CR-10,
2. Opened preferences and in printers settings added a new CR-10 printer (CR-10 #2).
3. removed the previous printer. Now I have only one printer
4. No warning messages. Everything is correct.
My assumption, I missed some steps in reproducing scenario. Or while Cura upgrading the setting files were corrupted. It this case I suggest to remove Cura setting files from _C:\Users\\<your username>\AppData\Roaming\cura\_

LipuFei, 2018-10-10:
@Liger0 It's hard to say what happened. Could you upload your cura directory so we can have a look? You can find it here
```
C:\Users\<your-user-name>\AppData\Roaming\cura
```

Phil67ago, 2018-10-12:
I also have similar problems with weird Profiles.
Seems that the way profile and current settings are structured is a little bit to complicated.
I think the idea behind this is that a change in one profile should propagate to all projects using that profile.
Nice idea, but if you don't know all project using a particular profile it might result in changes in old project which you don't like.

So my suggestion separate profiles and current setting totally.
This is what you should be able to do:
- Copy in profile to current project settings.
- Copy out current project settings to profile.
- Compare current project to profile. Copy in hole or selected items either way.

So here is one of the problem I had with profiles:
- Had one project with a linked profile.
- Change some of the projects settings.
- Created a new profile from project settings.
- Of some reason I get a #2 added to that profile name.
- Tried to remove (rename) the #2 because there where only one of that name.
- Got a number #3 instead. 
- Now I delete that because it seems not working.
- Close profile window.
- Now all project settings where updated from another profile.
- Open the linked profiles and see that Layer Height has changed from 0.1 to 0.2
- At same time Initial Layer height has changed from 0.2 to 0.1
I don't have that settings any where in my profiles.

As you see everything seems to be a total mess.
My suggestion is simplify profiles and setting and remove all heretical storage of settings and just
store profiles as a plain settings file, one for each profile.

alekseisasin, 2018-10-15:
"but if you don't know all project using a particular profile it might result in changes in old project which you don't like."
Every time when you load a project is triggered Version Upgrade which checks old project files and makes (required)changes according to the next Cura Version. This is an ideal scenario. Not ideal is exactly what you described if something goes wrong then rest is will be wrong. 
I agree with you that it should be much simpler, but while loading Cura needs to know different dependencies and they are stored hierarchically. 

We improve it, but it is a long-term issue.

Liger0, 2018-11-18:
Hi there. When working with the filament settings it appeared again. I will attach the filament settings profile, and tell me if anything more is needed.

[image]
[ABS.xml.zip](https://github.com/Ultimaker/Cura/files/2593229/ABS.xml.zip)

closed 2020-06-20 by no-response[bot]