Ultimaker/Cura #4500: [3.5.0] Segmentation Fault

thattolleyguy, 2018-10-06:
**Application Version**
3.5.0

**Platform**
Ubuntu 18.04

**Printer**
Printrbot Simple Metal

**Steps to Reproduce**
Open Cura

**Actual Results**
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number
Segmentation fault (core dumped)

**Expected results**
Cura should load normally

**Additional Information**

ChrisTerBeke, 2018-10-06:
We've see font issues on Linux before but were never able to really debug and fix them. Closing until re-appears.

closed 2018-10-06 by ChrisTerBeke

fieldOfView, 2018-10-06:
Note that the Fontconfig error being logged does not mean that that is also the cause of the segmentation fault. AFAIK, the fontconfig errors have been there since forever, and also happen on systems that do not have a seg fault.