Ultimaker/Cura #4503: The distance along the Z-axis does not match the number of layers

alexand1, 2018-10-06:
The distance along the Z-axis does not match the number of layers

**Application Version** - 3.5.0

**Platform** - Windows 10

**Printer** - Ultimaker 2

[image]

alekseisasin, 2018-10-08:
@alexand1 , Could you please share your project?

nallath, 2018-10-08:
Also, if you want to change the layer height, it's best to switch the profile altogether.

Ghostkeeper, 2018-12-05:
I'm unable to reproduce this issue. I loaded a cube of 40mm tall, and sliced it with 0.1mm layer height and 0.27mm initial layer height. This produced 398 layers, which is correct.

It must be model dependent.

closed 2019-01-08 by nallath