Ultimaker/Cura #4504: Printbot Simple Mettle Short Salling during plot on large models

TrackingTommy, 2018-10-06:
While plotting a 120 x 110 x 30 mm project with an I Mac and a Printbot Simple Mettle, the printer will  start  expreenceing short stalling about 1/8 to 3/16 into the plot.   If one pauses the plotter every 10 minuits or so and allow it to catch up to Cura 3.5.0 you get a good plot.  If you do not pause the printer from Cura 3.5.0 the unit stalls for a short perods of time, which the stalls get longer  as time passes,  adding more heat to the model at diffrent points and leaving distortions, if allowed to do this long enough, you can get a shift in the x & y axess and end up with spider webs and a mess 1/3 into the model plot.      At this point I will use 3.5.0 for very small projects and fall back to Cura 15.04 for the bigger models.   Details listed below:
-->

**Application Version**
3.5.0

**Platform**
I Mac with 2.7 GHz Intell Core I5 running macOS High Sierra version 10.13.6

**Printer**
Printbot Simple Mettle

**Steps to Reproduce**
Start the plot and watch

**Actual Results**
It appers that the Printbot Simple is getting more data than it can mannage and starts to have small stalls untill the x & y axixes gets messed up and then does not know whare the model is any more.

If the "pause" button is clicked wait 30 seconds and then click on "resume" and continue the process every 7 to 10 minuits then the plot will be OK

**Expected results**
Not to have to hover over the software and intervene with it in order to get a good plot.   Falling back to Cura 15.04 to do plots that take longer than 15 to 20 minuits to execute

**Additional Information**
Not having this issue with Cura Version 15.04.   

***Wish List***
It would be nice to have the manual maintance function back in 3.5.0 that was availble in version 15.04, whare you can manually set the table and extruder heat, check X, Y & Z access and check HOME operations.   Thes feature is very handy for preforming checks and maintance on the plotter.

Thank you for your time and kind consideration of this matter.

nallath, 2018-10-08:
Unfortunately, we don't support USB printing since all Ultimaker printers print from SD or network. Even if we would support USB printing, we still wouldn't be able to test any of this since we don't have a Printerbot.

TrackingTommy, 2018-10-08:
Thank you for getting back to me.

Printing USB from my iMac to the Printbot using the Cura 15.04 works great, so Ill keep using that for now.

Again thank you for getting back to me.

Tom Johnson
Gladstone MI

nallath, 2018-10-09:
You could also use Cura to slice and then send it to the printer by means of printrun. That way you can use the new slicing features.