Ultimaker/Cura #4505: 3.5 Slow Rendering on Mac Pro El Capitan

kevgee, 2018-10-06:
There 

**Application Version**
Cura 3.5 Update

**Platform**
Mac Pro 2009 running El Capitan

**Printer**
Kossell Delta 3D Printer - Custom

**Steps to Reproduce**
run the program, load an STL

**Actual Results**
Model view frame rate is very slow compared to older version of Cura

**Expected results**
Same frame rate or faster (Please!)

**Additional Information**
One STL to use is the KM_Rosebloom stl from yeggi.com

alekseisasin, 2018-10-08:
Hi @kevgee , how did you measure the framerate? I assume the issue related to rendering or video card.  It is quite specific. A similar issue was reported from some other Mac users. If is it possible can you update system drivers?

kevgee, 2018-10-08:
Hi, I measured the frame time response using my eyeballs and hands. The input time to rendering response is slow enough that you can feel it (order of a second for the image to update).
I have all the available system updates for the OS, I assume the driver is provided as part of the Apple updates. Would you advise going directly to NVIDIA instead for a display driver?

Thanks!

alekseisasin, 2018-10-09:
Unfortunately, I cannot advise because I do not use Mac, but you can try different options. Try to use direct NVIDIA drivers.

iNherjer, 2018-10-09:
I have the same problem, the image stucks for around a half second the you try to move or rotate the object or view point . Also scrolling through the settings is really slow and stuttering

iNherjer, 2018-10-09:
It’s an IMac 27“ end 2009 
Never had problems until the last update

alekseisasin, 2018-10-09:
We also did not have the same issue on testing Mac computers. For this reason we need to go more specific to computer configurations.  IMac 27, do you have some extra data, maybe you can find on internet link where is described you model.

iNherjer, 2018-10-09:
2.8ghz Core i7
8gb 1067 mhz ddr3
ARI hd 4850 512 mb 
Mac OS 10.13.1

That’s all I got.

mparson, 2018-10-10:
Seeing similar performance problems on my late 2009 27-inch iMac
2.8 GHz Core i7
16 GB RAM
ATI Radeon HD 4850 512 MB
macOS Sierra 10.12.6

alekseisasin, 2018-10-11:
May I ask you to turn off the internet connection and observe how Cura works in this case?

mparson, 2018-10-11:
I quit Cura.  Unplugged my ethernet, made sure wifi was still disabled, fired up Cura, loaded a model, and it was still super slow, like low single-digits of frames per second, if not seconds per frame.

alekseisasin, 2018-10-11:
When you start Cura can you check on mac which processes use most of CPU?

mparson, 2018-10-11:
Just sitting there, not doing anything, Cura's CPU usage is pretty low.  When I start trying to rotate the build plate Cura jumps up to 97-98%.

fraesoholic, 2018-10-16:
i have the same problem with Cura 3.5, 3.51 beta and 3.51.
Mac Pro 3.1, 2 x 3 GHz Quad-Core Intel Xeon, 32 GB 800 MHz DDR2 FB-DIMM, mac OSX 10.11.6 (15G22010)

[In this screencast you can see how slow everything is](https://www.screencast.com/t/U3wkediNbB8i), the menu structure can even be interrupted if I do something else. As soon as I perform any action, the processor load display jumps to almost 100%.
I have reinstalled several times, deleted library folders, interrupted network without any changes.
In the attachment stderr.log.zip and cura.log.zip
Which log files are still needed to narrow down the error?

It is not the computer's fault, if I start in Bootcamp under Windows 7  the Windows version runs very fast and smoothly.
[stderr.log.zip](https://github.com/Ultimaker/Cura/files/2483712/stderr.log.zip)
[cura.log.zip](https://github.com/Ultimaker/Cura/files/2483713/cura.log.zip)

++ ralf

pieri70, 2018-10-25:
Same issue here
Cura 3.5.1 slow in everithing. Layer view is almost impossible to use and CPU runs at 100%..
MacBook Air mid 2011. Core I5 4GB Ram El Capitain.
I use ideamaker, Slic3r PE S3D and they work good

bleft, 2018-11-03:
Me 2. iMac late 2010 with ATI Radeon HD 4850 512 MB.
3.4 runs like a charm.

pieri70, 2018-11-15:
Again with cura 3.6.0 macbook air mid 2011 I5 1.7Ghz 4GB Ram, 256SSD with mac os high sierra just freshly installed Cura is unusable.
In Solid view if I don't do anyting CPU stays at 3-4%. as I click anywhere it jumps @150% making any selection/option really slow.
In layer view once the stl is sliced CPU is 120% all the time.

pieri70, 2018-11-17:
Again, this problem is occurring in macbook air as above.
I tried to install 3.4.1 and works perfectly.
here I posted the cura log
https://github.com/Ultimaker/Cura/issues/4820

boboxx, 2018-11-29:
Cura 3.6 is extremely slow and is unusable OSX 10.13.6 (Mac Pro early 2009 2x 2.26GHz Quad-Core Xeon with 12GB ram on SSD) I also tried Cura 3.41 as mentioned above and work perfectly!

[cura.log](https://github.com/Ultimaker/Cura/files/2630842/cura.log)

Altairama, 2018-12-08:
Seems like a common problem with the new version, I´m having the same issues in mine.
(MacBook pro 7.1, core2duo 2.4Ghz, 8Gb ram, SSD With the latest version of high sierra)
Was working perfectly fine with the version 3.4
I´ve upgraded for mavericks to be able to use the new 3.5-3.6 But as you can imagine if in a quad core i7 is extremely slow with the core 2 duo is completely unusable (the bigger and more complex the model is the worst).
The CPU ussage is extremely high.

ubermacin, 2018-12-21:
Another old(ish) mac user, zero problems until 3.5, 3.6 and 4.0 beta run like poo.  Exact same config as previous user

captainstarfish, 2019-01-06:
Another one. On fresh startup it's ok - not great, but ok. Add a model and slice, it starts bogging down. Switch from Solid view to layer view and I might as well kill the app and start again, looking at 5s per frame refresh on all but the smallest models.

Running High Sierra (10.13.6) on a late 2009 27" iMac, 2.8GHz i7, 32GB RAM, SSD and ATI Radeon HD 4850 512MB.

fieldOfView, 2019-01-07:
In the general preferences, set the option to Force compatibility mode in layer view.

captainstarfish, 2019-01-07:
> In the general preferences, set the option to Force compatibility mode in layer view.

THANK YOU!!!

It's still a little sluggish but so, so much better now than it was before: it's actually usable again now. Sweet!

fieldOfView, 2019-01-07:
It looks like with the currently used Qt version, OSX will create a software-backed OpenGl 4.1 context, instead of failing to create that context. This means all of Cura will draw with software rendering instead of being hardware accelerated.

UM.View.GL.OpenGLContent.detectBestOpenGLVersion() will need to check for software renderers.

LipuFei, 2019-01-09:
JIRA ticket: CURA-6092

pieri70, 2019-01-09:
I can confirm that 3.6 with Force compatibility mode in layer view is a little better in MacBook AIR mid 2011 High Sierra

fieldOfView, 2019-04-11:
I have created a repackaged build of Cura 4.0 with my fix for the slow rendering on Macs that don't support the newer OpenGL features that Cura uses for the advanced layer view:

http://files.fieldofview.com/cura/Ultimaker Cura-4.0.0.dmg

Other than the change in the OpenGL context, it should be axactly the same as the original build, and work with your existing configuration.

Please let me know if the build works, and if it is indeed faster.

captainstarfish, 2019-04-11:
Hi Aldo

Thanks for doing this!

Unfortunately the package is coming up as corrupt after download. Perhaps something to do with Apple’s mad code signing rules?

Cheers

Simon

fieldOfView, 2019-04-12:
I have replaced the build, could you try again? Same url:
http://files.fieldofview.com/cura/Ultimaker%20Cura-4.0.0.dmg

pieri70, 2019-04-12:
Super!
It works fine, both in conpatibility mode and in normal one.
I loaded a stl file, pan and zoom are smooth and "fast" (normal for my MB Air mid 2011 I5 4GBRAM SSD).
There is no freezing view both in prepare and in preview mode.

Do you think this mod can be integrated into official release?

Thanks!

fieldOfView, 2019-04-12:
Thanks for testing!

> Do you think this mod can be integrated into official release?

For all intents and purposes, this is the fix integrated into an official release.

Ultimaker are not likely to release a Cura 4.0.1 with this fix. If my fix does not cause problems on other systems, it will be part of Cura 4.1.

captainstarfish, 2019-04-13:
OMFG

Thank you, Aldo - you just fixed the one thing that made me dread using Cura in spite of all its excellent features.

Time to play with printer settings!

Cheers

Simon

fraesoholic, 2019-04-13:
> If my fix does not cause problems on other systems, it will be part of Cura 4.1.
+1

The official version 4 ran on my Mac Pro with high sierra only very badly and very slowly. Your version runs very well, without any jerking and fast.
Thanks a lot for that.

ralf

nallath, 2019-04-16:
The code has been merged into master. We (the devs) tested it on a handfull of systems and it didn't cause any issues. It still needs to pass through our QA department, but as it stands it's really likely to be in 4.1.

closed 2019-04-16 by nallath

Fvrajo, 2019-05-09:
> I have replaced the build, could you try again? Same url:
> http://files.fieldofview.com/cura/Ultimaker%20Cura-4.0.0.dmg

Great thanks, is working

ubermacin, 2019-06-19:
Checkin in here to say thank you!  4.1 downloaded and runs like the old Cura (that is to say, really nice on a 9 year old MacBook pro).  I really appreciate it!