Ultimaker/Cura #4506: Cura.exe gets stuck before splash screen.

githubcura35, 2018-10-06:
**Application Version**
Cura 3.5

**Platform**
Win 10 1803

**Printer**
Creality Ender 3

**Steps to Reproduce**
Install Cura 3.5 from scatch and start
[Configure Seettings for Ender 3](https://youtu.be/umulfK775i0)
Reboot
Start Cura 3.5

**Actual Results**
Cura.exe gets stuck before splash screen.
The only way I have found to fix that is to deinstall and reinstall 

**Expected results**
Splash screen should come up and Cura should start

**Additional Information**
Cura 3.4.1 works just fine
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/2453282/DxDiag.txt)

alekseisasin, 2018-10-08:
It is very strange.
If you started Cura at least one time then the log file cannot be empty.
Where do you install Cura? Do you have enough free space on your hard drive?
Before installing Cura remove all Cura settings from _%APPDATA%\cura\<Cura version>\_

LipuFei, 2018-10-08:
Hi @githubcura35 @Kernelhacker , could you check if there is a file/files called "uranium.lock" or "*.lock" in the folder `C:\Users\<your-user-name>\AppData\Roaming\cura\3.5`? If so, could you try to delete them and see if it works?

Kernelhacker, 2018-10-08:
> Hi @githubcura35 @Kernelhacker , could you check if there is a file/files called "uranium.lock" or "*.lock" in the folder `C:\Users\<your-user-name>\AppData\Roaming\cura\3.5`? If so, could you try to delete them and see if it works?

Sorry, in my installetion there is no lock-file in the 3.5 folder.

Kernelhacker, 2018-10-08:
> 
> 
> It is very strange.
> If you started Cura at least one time then the log file cannot be empty.
> Where do you install Cura? Do you have enough free space on your hard drive?
32GB remaining on that partition.
> Before installing Cura remove all Cura settings from _%APPDATA%\cura<Cura version>_
I removed the cura3.5 folder and reinstalled Cura 3.5. On the first start I was asked for sending anonymous slicing info and I agreed. Now when I save a sliced model, cura crashes/closes. The last messages from stderr.log are:
2018-10-08 19:24:37,252 - DEBUG - [MainThread] UFPWriter.UFPWriter._createSnapshot [30]: Creating thumbnail image...
2018-10-08 19:24:37,268 - INFO - [Thread-3] SliceInfoPlugin.SliceInfoJob.run [29]: Sending anonymous slice info to [https://stats.ultimaker.com/api/cura]...

I can restart cura atm., but I have not yet tried to reboot. I'll add that information later.

LipuFei, 2018-10-08:
@Kernelhacker Could you send us your `%APPDATA%\cura` folder, after your first success run but before you start it the second time?

Kernelhacker, 2018-10-09:
> 
> 
> @Kernelhacker Could you send us your `%APPDATA%\cura` folder, after your first success run but before you start it the second time?
Here you go:
[3.5 - FirstStart.zip](https://github.com/Ultimaker/Cura/files/2461735/3.5.-.FirstStart.zip)

The very interesting thing is: I can start cura 3.5 multiple times without a problem. It only fails after a reboot after the installation!

LipuFei, 2018-10-09:
@Kernelhacker Hmm... I wonder if you reboot and then start Cura, about how long does it get stuck (> 1min for example)? When that happens, could you force quit Cura and upload the same folder again? I thought it only works the first time, so the first time log file and data files seem interesting. Thanks.

Kernelhacker, 2018-10-09:
> @Kernelhacker Hmm... I wonder if you reboot and then start Cura, about how long does it get stuck (> 1min for example)? When that happens, could you force quit Cura and upload the same folder again? [...]
There you go:
[3.5 - after getting stuck.zip](https://github.com/Ultimaker/Cura/files/2461813/3.5.-.after.getting.stuck.zip)

Kernelhacker, 2018-10-10:
> Hi @Kernelhacker , it all seems fine... I'm not sure what went wrong there. Maybe you could try a Cura 3.5.1 candidate build for us? 

No change - it still hangs on startup after a reboot. Maybe it has something to do with Windows Preloading of DLLs? Or the installer leaves something undone, that is completed after a reboot? 

> There are at least some fixes for Ender 3. Thanks in advance.

Hm, I did not plan to buy an Ender 3 ;)

LipuFei, 2018-10-10:
This is so weird... The installer should have done its job. Basically it only needs to copy some files. It works for you before you restart, so I think the files should be in place already. After you restart and before start Cura again, if you open the "Task Manager" to show all processes, do you see any Cura processes running? such as "Cura.exe"

Kernelhacker, 2018-10-10:
> This is so weird... The installer should have done its job. Basically it only needs to copy some files. It works for you before you restart, so I think the files should be in place already. After [...]

No cura* processes are running after a reboot. It is weird, indeed. Funny thing is - until now it seems only githubcura35 and me are facing that problem. I just see - I'm running a windows insider preview here. Maybe that is the problem.

LipuFei, 2018-10-10:
> No cura* processes are running after a reboot. It is weird, indeed. Funny thing is - until now it seems only githubcura35 and me are facing that problem. I just see - I'm running a windows insider preview here. Maybe that is the problem.

That's a possible explanation... Could you try 3.5 on another Windows you can find?

mod25, 2018-10-11:
@LipuFei your beta worked for me in this issue. thank you. do you provide also beta for mac which have the ender 3 profile fixed?
thank you very much

mod25, 2018-10-13:
@LipuFei sorry to tell you that today i'm not able to start cura same issue as before.

githubcura35, 2018-10-17:
Same after upgrade to Cura 3.5.1. Only workaround is to start cura.exe in win8 compatibility mode.

lexter12, 2018-10-21:
I have the same exact problem (Cura doesn't start - doesn't even display the splash screen, nothing in the logs, cura.exe shows up in task manager after running but is doing nothing). I have Cura 3.5.0 on standard Windows 10 pro (10.0.17134) 64bit. I had the same issue with 3.4.x also.
Running Cura in Windows 8 compatibility mode does work though!

alekseisasin, 2018-10-22:
@lexter12 , Can you do one thing for me. When you start Cura and it does not even display the splash screen, can you open the _C:\Users\\AppData\Roaming\cura<Cura version>_ folder and check whether do you have a _cura.lock_ file in this folder or not?

alekseisasin, 2018-10-22:
@lexter12 , Can you do one thing for me? When you start Cura and nothing happens (doesn't even display the splash screen), open 
_C:\Users\\AppData\Roaming\cura<Cura version>_ folder and check whether do you have _cura.lock_ file or not?

lexter12, 2018-10-22:
I checked as I saw it mentioned above, there is no .lock file and logs are old/empty (didn't even change modification date upon run). And again - troubleshooting compatibility and changing to run as Windows 8 works...
I can see Cura.exe in the task manager, it's consuming 40M memory, has 0 CPU and that's it.

trunneml, 2018-11-01:
Same here. Windows 10 (no insider) Cura starts after installation, but not after a reboot. Tried 3.5.0 and 3.5.1.

trunneml, 2018-11-03:
I tried 3.6 beta to day. Same Problem. Won't show splash screen. Running Cura in Windows 8 compatibility mode helps.

aeozyalcin, 2018-11-08:
Also having the same issue with Cura 3.5.1. I am running Windows 10 RS5, older versions launch just fine...

knopserl, 2019-01-04:
I'm running Windows  10 x64 (1803). For me Cura 3.6 and 4.0 beta gets stuck on splash screen at loading preferrences (can wait up to an hour). Application no longer responding. delete of config file does not help. Logfile does not really contain useful info's.
Cura 3.4.1 works fine.
After reading the post of @trunneml, I also tried to set Windows 8 compatibility mode (on Cura.exe), suddenly in runs!

val742, 2019-04-20:
> Hi @githubcura35 @Kernelhacker , could you check if there is a file/files called "uranium.lock" or "*.lock" in the folder `C:\Users\<your-user-name>\AppData\Roaming\cura\3.5`? If so, could you try to delete them and see if it works?
It seems that this helps now with the 4.0 again, just look for the .lock file and remove it to open Cura 4.0 again.

Ghostkeeper, 2019-04-24:
The lock timeout still works fine for me in 4.0.