Ultimaker/Cura #4508: lost printer profile

stilgarhammer, 2018-10-06:
**Application Version**
2.5.0

**Platform**
windows 7

**Printer**
prusa i3

**Steps to Reproduce**
close cura

**Actual Results**
reopen cura and printer profile and all settings gone

**Expected results**
no issues, all settings intact

**Additional Information**

ChrisTerBeke, 2018-10-06:
2.5.0? That's very old, we're not supporting it anymore.

stilgarhammer, 2018-10-06:
everything was working good, I closed it last night. open it this morning and all settings gone. is there a file i can fine to restore those setting?
I liked 2.5.0

LipuFei, 2018-10-08:
Hi @stilgarhammer , you can have a look at your configuration folder, which should be located in `C:\Users\<your-user-name>\AppData\Roaming\cura`. Unfortunately, we have officially ceased support for Cura versions before 2.7. Could you try 2.7 or something?

alekseisasin, 2018-10-12:
@stilgarhammer , did you have recent updates in windows? On the internet a lot of articles that Windows 10 update might remove personal data. For Windows 7 it could be also.

Ghostkeeper, 2018-12-19:
We can't really help you with this. We have not enough information to reproduce the issue and it's very unlikely that it reproduces in version 4.0. We're not going to release a new version of 2.5.

closed 2018-12-19 by Ghostkeeper