Ultimaker/Cura #4509: Cura 3.5 at end of print disables my steppers? without m18 or m84 in code?

undertowx, 2018-10-07:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

nallath, 2018-10-08:
If there is no g-code telling it to do so, it's not Cura that did it ;)

undertowx, 2018-10-08:
in machine settings I had rep rap(volumetric) for my older Cura profile.
In cura 3.5 there was no Volumetric rep rap setting to I chose Marlin (volumetric.)
Could that be the proublem?

nallath, 2018-10-09:
Maybe. I'm not that familiar with non UM printers.

Ghostkeeper, 2018-12-07:
Cura can only choose what g-code to send to the printer. If the g-code is all right but the printer is doing something strange, then Cura can't help you out.

Let us know if there is something that Cura can put in the g-code to prevent the stepper motors from losing power. Maybe we'd have to put `M28` at the end of the print? If your home location doesn't involve the print head moving through the build volume where the print could be, that is... And which printer are you using?

For now, this appears to not be a problem in Cura but in your printer.

closed 2018-12-07 by Ghostkeeper