Ultimaker/Cura #4511: Move extruder to safe position, related to #3451

Tinchus2009, 2018-10-07:
**Application Version**
<3.5 64 bit>

**Platform**
<win10 64B>

**Printer**
<custom prusa i3>

**Steps to Reproduce**
<I create a dual color model, then slice and run the print>

**Actual Results**
<In each tool change, teh tool stays OVER the model while reaching the printing teperature
. I have looked for info, and I found issue #3451 . This issue is makrked as "SOLVED" and it is closed. I read the articled but I dont see a final solucion posted there. Is there a solution?>

**Expected results**
The extruder should move to the prime tower and warm up there

**Additional Information**

Tinchus2009, 2018-10-08:
@amagr0 were you able to solve it?

AbeFM, 2018-10-20:
It should move to ABOVE the tower, not on it. The nozzles can melt into any part they rest over (worse if the temp is dropping) - and when it moves it can take the part with it.

Ghostkeeper, 2018-12-27:
The solution is the same though: Use the start/end positions to set the resting location:

    "machine_extruder_start_pos_abs": {"default_value": true},
    "machine_extruder_start_pos_x": {"default_value": 0},
    "machine_extruder_start_pos_y": {"default_value": 0},
    "machine_extruder_end_pos_x": {"default_value": 0},
    "machine_extruder_end_pos_y": {"default_value": 0},

You can't set a Z coordinate here though or require a z-hop. I suggest you put it somewhere next to the prime tower or on the inside.

If you didn't create the printer definition yourself, it's the author of that definition that made the mistake. You can alternatively download the Machine Settings plug-in from the Toolbox/Marketplace to change these settings a bit easier from the Cura interface.

closed 2018-12-27 by Ghostkeeper

fundamental4, 2019-03-04:
Hi all,
Beginner user. I need help.

I have the same problem and it’s not solved for me, or at least I don’t know how to do it.
(Ver. Cura 3.6.0)

I agree, the heating shouldn’t be in the prime tower because it can take the part with the nozzle, as it is a repeated operation. Also, the extruder could be stagnant if the quantity of filament dropped is big.

Even if the warm up time is OUT of the part and tower, it'd be fine if once the heating is finished, the nozzle travels to the prime tower to be cleaned.
I guess, in the same way I can decide the coordinates of Prime Tower, I could decide the waiting time position.

What can I do to change the position while heating?
How/Where can I modify the settings to solved the problem as Ghostkeeper?

Thank you

Ghostkeeper, 2019-03-05:
Hey @fundamental4, this problem was inside the printer definition. So it's not your fault, but rather a mistake made by the person who created that profile. Which printer are you using?

> How/Where can I modify the settings to solved the problem as Ghostkeeper?

That would be in the printer definition. Modifying that is considered expert-level though, as it involves modding Cura. You'd need to go into Program Files (or the package if not using Windows), finding the resources folder, and under definitions modify the .def.json file that is used by your printer. You could add the settings I posted there. Not for beginner users though, I'd say. But if anything goes wrong, re-installing Cura should work fine and even let you retain all your settings.

fundamental4, 2019-03-05:
Hello Ghostkeeper,
thank you for your answer, because I realized you closed this thread on 27dec2018.

Mine is a kit printer, "MendelMax", made by **makergal.es**.
I'll speak with them.