Ultimaker/Cura #4512: [3.5.0] Changes in filament diameter do not take effect in gcode

r011e, 2018-10-07:
I recently downloaded Cura for Linux 3.5.0 from the Ultimaker website.
The problem was that the printer did not extrude enough filament despite changing the diameter settings.

The problem is that the changes to the filament settings don't take effect in the gcode. If you change the diameter it should change the values for the extruder in gcode

The problem can be reproduced like this:
1. Change diameter size of material (and printer settings)
2. Prepare model and save file
3. Repeat step 1 and 2 with different settings
4. Compare the value of the extruder in gcode

I tested version 3.4.1 and it changed the values for the extruder as expected

LipuFei, 2018-10-08:
For dev, JIRA ticket: CURA-5799

LipuFei, 2018-10-08:
PR: https://github.com/Ultimaker/Cura/pull/4521

LipuFei, 2018-10-09:
Fixed in #4521

closed 2018-10-09 by LipuFei