Ultimaker/Cura #4513: level bugs

muratug, 2018-10-07:
hello program giving bugs as you see at the pictures

**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

muratug, 2018-10-07:
[denizyıldizi.zip](https://github.com/Ultimaker/Cura/files/2454103/denizyildizi.zip)
cura 3.5.0
anet a8 as you see

nallath, 2018-10-08:
I don't see any image attached. Can you please fill in the bug template so that we know what the actual problem is?

muratug, 2018-10-08:
[image: denizyıldızı1.jpg]

Jaime van Kessel <notifications@github.com>, 8 Eki 2018 Pzt, 17:56
tarihinde şunu yazdı:

Ghostkeeper, 2018-12-07:
The provided model seems to slice fine (at least with the Ultimaker S5 profile; I don't have your Anet A8 profile). We can't reproduce any bugs that you have. I'm closing this issue because there is no information to figure out what the problem is.

closed 2018-12-07 by Ghostkeeper