Ultimaker/Cura #4515: Cura 3.5 sequential batch file printing

elbertoman, 2018-10-07:
**Application Version**
3.5.0 x64

**Platform**
Windows 

**Printer**
Not relevant

**Steps to Reproduce**

1. Load any model and multiply, nest if necessary
2. Shift + Select in reverse order the sequence in which you wish the batch of models to be printed
3. Group the models
4. Ungroup the models
5. Slice
6. View in layer mode

**Actual Results**
Cura determines the order in which to sequentially print the batch of models

**Expected results**
Cure prints the models in the reverse sequence in which they were selected and then grouped together.  This was the behavior in Cura 3.4.1

**Additional Information**
The batching for sequential model printing was superior in Cura 3.4.1 because it allows the user to manually determine the order in which all the models are printed and optimize for gantry collisions and overlaps.  Please enhance or roll-back to Cura 3.4.1 behavior.

nallath, 2018-10-08:
This was never intentional behavior. Any changes in print order due to grouping / ungrouping was emergent behavior. 

As for gantry colissions, how is this an issue when printing all at once? Even when printing one-at-a-time, Cura should also resolve that by itself (thus ensuring that you don't have to rely on you making the correct decisions)

elbertoman, 2018-10-09:
The ability to manually determine a print path per object for sequential printing is incredibly important for batch manufacturing processes and also why all the more advanced software platforms by Stratysys and 3D systems support it.  Cura is not smart enough yet in 3.5 to optimize for build plate space/nesting because it predicts gantry collisions that will not happen due to the sequential nature of printing.  So again, the behavior in 3.4.1, even if it was emergent, was superior to what is currently happening in 3.5.

Ghostkeeper, 2018-12-13:
It is not our intention to implement manual selection of the printing order. Instead, we'd like Cura to determine a good order automatically, so as to not burden the user with this task.

The algorithm is not perfect and needs improving, but we won't implement a manual ordering workflow yet to keep it simple to use.

If we had intended to allow for a manual ordering of nodes, we would've made it easier to use than by grouping/ungrouping, and more obvious to find and see what the result is. But that's not on our planning right now.

closed 2018-12-13 by Ghostkeeper