Ultimaker/Cura #4518: release 3.5 not on github?

burdickjp, 2018-10-07:
Why isn't release 3.5 available under "releases" on github?
My package manager pulls from github.

wilriker, 2018-10-08:
The same already happened with previous release (see https://github.com/Ultimaker/Cura/issues/4019). It seems this is very delayed.

nallath, 2018-10-09:
Because we're very good at forgetting this

burdickjp, 2018-10-10:
You can take comfort in knowing that you can forget this and we'll remind you every time.

Ghostkeeper, 2018-12-13:
It's not so much the forgetting but more that I'm lazy and slightly worried that we might hit a file upload limit some day, so I'm trying to only upload one release per minor version increase. If I see a patch release coming up I'll wait for a bit to upload the version with the patch.

Anyway, 3.5.1 is uploaded now so this is fixed.

closed 2018-12-13 by Ghostkeeper