Ultimaker/Cura #4519: CURA 3.5 Closes when trying to add ant printer

LouieRodotech, 2018-10-08:
**Application Version**

Cura 3.5
**Platform**

Win 10
**Printer**

JG Aurora A5
**Steps to Reproduce**

Open program add printer, select printer to configure application closes. 
**Actual Results**

Application closes
**Expected results**

Setup bed size
**Additional Information**

[image]
 works fine in ver 3.4

Appesteijn, 2018-10-08:
Adding a custom FFF printer with that names works fine on my Mac. Can you post your logfiles?

LipuFei, 2018-10-08:
Hi @karabas , for you JG Aurora A5, did you follow the steps on their website? For example, copying the *.json files to the latest Cura?

Link: https://jgaurorawiki.com/a5/printing-profiles

nallath, 2019-04-04:
Since no more information was provided, I'm closing this issue.

closed 2019-04-04 by nallath