Ultimaker/Cura #4523: [3.5.0] Crashes when trying to load gcode files.

erispe, 2018-10-08:
**Application Version**

3.5.0

**Platform**

Windows

**Printer**

Original Prusa i3 MK3

**Steps to Reproduce**

Either drag-n-drop a .gcode file into the main window or using File > Open File(s)

**Actual Results**

Program exits to desktop.

**Expected results**

Model should load into the scene. I can load the same gcode files without issue in 3.4.1 and it does not matter if the gcode was generated in 3.50 or 3.4.1, they all work in the previous version and all crash in the newer.

**Additional Information**

[cura.log](https://github.com/Ultimaker/Cura/files/2457207/cura.log)

LipuFei, 2018-10-08:
Hi @erispe , I don't see anything interesting in the log file. If possible, could you send that gcode file you were trying to load? Thanks

Drayson80, 2018-10-09:
Hi! I experience the same issue. 
Cura 3.5
Win10
UMO
same issue description as from erispe...
Gave it a try with very old gcode files and newly sliced gcode. Log-file looks good, no indication also on my side.

nallath, 2018-10-09:
If you don't drag them in but load them via file-> load, does then the same happens? Also, could you share the g-code where this happens?

Lundsgaard2, 2018-10-09:
Same problem here, with a twist.
When I open files i created from scratch using sketchup there is no problem.
Printing items from 3dwarehouse makes Cura 3.5 crash.
It works in earlier versions.

LipuFei, 2018-10-09:
Hi @Lundsgaard2 , if possible, could you share the file(s) that don't work for you? Thanks.

Lundsgaard2, 2018-10-09:
Here You are.

[DoNotWork.zip](https://github.com/Ultimaker/Cura/files/2460133/DoNotWork.zip)

LipuFei, 2018-10-09:
Hi @erispe @Lundsgaard2 , I see that you are using Prusa i3 MK3. Please check https://github.com/Ultimaker/Cura/issues/4497 that we have found the printer profiles provided by prusa3d for MK2S and MK3 don't work in Cura 3.5. Could you check if that's the case for you? If you switch to use a different printer (one that's available in the "Add Machine" list), does this problem still occur?

Drayson80, 2018-10-09:
ok, sorry for late reply  @nallath 
I playd a bit around and it seems that my issue is linked to the SD-card reader....
If I open an old gcode from my SD archive, Cura crashes. 
If I copy the same file to my local disk, it opens correctly...
Attached a file sliced 14/07/2017 with Cura_SteamEngine 2.6.1 for my UMO where I experienced the issue. (just rename it to .
[image]
gcode)

LipuFei, 2018-10-10:
@szepeu From the log file you uploaded, it crashes on OpenGL. It may have something to do with your graphics card driver. Looks similar to https://github.com/Ultimaker/Cura/issues/4302 but crashes in different ways. Could you update your graphics card driver? If you opened more than one Cura window at the same time, could you try just one? Thanks

Lundsgaard2, 2018-10-10:
Hi LipuFei. I am using AnyCubic I3 Mega and not  Prusa i3 MK3.
[image]

LipuFei, 2018-10-10:
Hi @Drayson80 , it sounds like the same problem as https://github.com/Ultimaker/Cura/issues/4502 and https://github.com/Ultimaker/Cura/issues/4495 (fail to open files on Windows on a remote/portable drive), which has been fixed. It will be in the next 3.5 patch release.

LipuFei, 2018-10-10:
@erispe @Lundsgaard2 Could you try opening files that's on your local hard drive, for example `C:\`, and then from some other drive? Does it make a difference?

LipuFei, 2018-10-10:
Hi @szepeu , sorry that ignore my previous comments... I have too many log files and I think I looked at the wrong one...

Lundsgaard2, 2018-10-10:
Ok

Venlig hilsen

Lars Lundsgaard Andersen
Faglærer
la@eucnord.dk<mailto:la@eucnord.dk>

EUC Nord, M. P. Koefoeds Vej 10, 9800 Hjørring
Tlf. 7224 6000  -  www.eucnord.dk<http://www.eucnord.dk/>

LipuFei, 2018-10-10:
Hi @erispe  @Lundsgaard2 @szepeu @Drayson80  , could you try a Cura 3.5.1 candidate build for us? We have a fix for #4495 and wonder if it works for you. Thanks!
Link: https://we.tl/t-0gqXweB9KG

Lundsgaard2, 2018-10-10:
I can`t install.

Venlig hilsen

Lars Lundsgaard Andersen
Faglærer
la@eucnord.dk<mailto:la@eucnord.dk>

EUC Nord, M. P. Koefoeds Vej 10, 9800 Hjørring
Tlf. 7224 6000  -  www.eucnord.dk<http://www.eucnord.dk/>

Lundsgaard2, 2018-10-10:
> Hi @Lundsgaard2<https://github.com/Lundsgaard2> , could you provide more specific information on why you can't install? Is it that nothing happens when you double click the installer, or the installation fails at some point, etc.?

Now it works. I can load my “donotwork” and save as gcode.
The printer is busy at the moment but I will test it later.

Venlig hilsen

Lars Lundsgaard Andersen
Faglærer
la@eucnord.dk<mailto:la@eucnord.dk>

EUC Nord, M. P. Koefoeds Vej 10, 9800 Hjørring
Tlf. 7224 6000  -  www.eucnord.dk<http://www.eucnord.dk/>

LipuFei, 2018-10-10:
@Lundsgaard2 Great to hear. I guess the problem is fixed for you. You can try 3.5.1 once it's released.

szepeu, 2018-10-10:
@LipuFei it does work. However, after regular Cura upgrades since 3.0, minor and major bugs and issues, I'm gonna stay with 15.06 - waaaaaay better performance and actually better gcode quality.

LipuFei, 2018-10-11:
I'll close this issue then.

closed 2018-10-11 by LipuFei

Lundsgaard2, 2018-10-11:
Thanks for Your rapid solution ☺

Venlig hilsen

Lars Lundsgaard Andersen
Faglærer
la@eucnord.dk<mailto:la@eucnord.dk>

EUC Nord, M. P. Koefoeds Vej 10, 9800 Hjørring
Tlf. 7224 6000  -  www.eucnord.dk<http://www.eucnord.dk/>

erispe, 2018-10-12:
@LipuFei, Wow! I forgot to come back this thread for a few days. Was just about to upload a gcode file for you to test and instead you already supplied a new build that fixes the issue. I am truly amazed!