Ultimaker/Cura #4524: Ironing misses some areas Cura 3.5.0

printingotb, 2018-10-09:
**Application Version**
3.5.0

**Platform**
Win7

**Printer**
Wanhao Di3

**Steps to Reproduce**
See attached project, I have seen this issue on several models but some simple models does not have the issue with the same settings.

**Actual Results**
Some areas are ignored by the Ironing function

**Expected results**
It worked with the same settings in all earlier versions of Cura3.

**Additional Information**
[image]
[image]

printingotb, 2018-10-09:
Forgot the project file:
[ironing issue.zip](https://github.com/Ultimaker/Cura/files/2460088/ironing.issue.zip)

Tomaskom-cz, 2018-10-15:
I encountered the same problem. Using Cura 3.4.1 or earlier, everything was fine. Now many tighter spaces get missed by ironing. Attaching a screeshot from gcode wiever (3.5 up, 3.4.1 down) where the missing ironing moves are clearly visible as lighter areas without the extra tight ironing lines. Some areas just get skimmed through with a single ironing line, others are omitted completely.
The two gcodes were produced using exactly the same project file and settings. 
[image]

Tomaskom-cz, 2018-10-24:
Just an observation than might help identify the issue: 
It seems the ironing malfunctiones exactly only in areas, where the individual ironing pattern moves (I'm using the default - ZigZag) would be shorter than a certain distance. That distance is consistently something around 6-7mm in my sliced models.

rburema, 2018-10-25:
Devs: CURA-5855

zaxhalmon, 2018-11-16:
Still an issue in 3.6, Didnt have this issue until i updated to 3.5.1. but still had it even after going back to 3.5.0. Using the same model in the past that i never had ironing issues on, I am still getting the above issue. It can be seen in Cura.

rburema, 2018-11-16:
@zaxhalmon We couldn't get this fix in before 3.6, the ticket is still open. We hope to adress this in the current sprint and/or near future.

theuncle, 2018-12-08:
I've noticed that putting the infill in giroyd the bug disappears and I can use ironing with zig zag pattern without issues

smartavionics, 2018-12-08:
Hello @theuncle. You are correct because the bit of Cura that is faulty breaks not only the ironing but also the gyroid infill (and zig-zag infill, sometimes). So I cunningly disabled the naughty code when the gyroid infill is used.

Ghostkeeper, 2018-12-21:
This is fixed in version 4.0. Thanks for the report!

closed 2018-12-21 by Ghostkeeper

printingotb, 2018-12-21:
Thanks a lot! :)

printingotb, 2019-01-01:
@Ghostkeeper Was this fixed in the Beta or after the beta? I still have the issue with Ironing in 4.0 beta (downloaded today).

I also found the following related issue when helping a guy out:
If I choose zigzag for bottom pattern initial layer, the bottom pattern does not fill the entire area.
Zig zag for the top/bottom layers also fails to fill in all areas (see for example layer 20 in the included project file).
Lines and concentric works as intended. 
Zig zag in combination with gyroid infill works as a fix, as it did with the ironing issue.
[image]
[ZigzagIssue.zip](https://github.com/Ultimaker/Cura/files/2719268/ZigzagIssue.zip)

Cheers!

kris2340k, 2019-01-02:
[image]

Printed nine of these squares today, 4 in cura 3.6 and after seeing this post 5 more. This is NOT FIXED.

Ghostkeeper, 2019-01-02:
@printingotb This was not fixed in the beta yet. It was fixed by [this merge](https://github.com/Ultimaker/CuraEngine/pull/911), about a week after releasing the beta.

We're planning to release a second beta for 4.0 soon, so you can still test it. Due to the UI change we allowed for a longer beta period this time.

printingotb, 2019-01-02:
Nice!
Just wasn't sure if you where aware that the bug affected the top/bottom lines as well.

rburema, 2019-01-09:
A thing to keep in mind: We'll be releasing a second beta before the 'final' (because of the big UI changes), so please keep an eye out for this issue. Like people noticed, this bug and it's fixes have it's tendrils in a lot of places.

betavr, 2019-03-02:
Chuck is advertising this broken feature on youtube: https://youtu.be/gh5wC4Ti95s

@Ghostkeeper are you guys going to release a new 4.0beta at last? You promised a new build in January man, and it's March already.

smartavionics, 2019-03-02:
There was a new beta released this week

https://community.ultimaker.com/topic/26773-ultimaker-cura-40-beta-2/

betavr, 2019-03-02:
@smartavionics But there is no binary. It's certainly not here: https://ultimaker.com/en/products/ultimaker-cura-software/list

smartavionics, 2019-03-02:
Hmm, you're right. Well, I don't know what happened there. I don't actually use those releases myself as I make my own. Sorry to confuse things.

fieldOfView, 2019-03-02:
The .../cura-software/list page incorrectly shows the release date of the first beta, but links to the second beta. Beta 1 is no longer available, beta 2 is.

smartavionics, 2019-03-02:
Thanks Aldo.

betavr, 2019-03-02:
@fieldOfView Good to know, thanks.

The fact that the binary isn't listed on the releases page (https://github.com/Ultimaker/Cura/releases) doesn't help either.

Tomaskom-cz, 2019-03-05:
Thank you, I think the release of the second beta should be better advertised. I was waiting for it and if I didn't comment on this issue, I would never know it's actually out. 
I can confirm the download page entry gets me a file named "Cura-4.0.0-BETA-2.AppImage" :)

rburema, 2019-03-05:
We've never had a 'second beta' before, so there's definitely some things we can improve in that regard if we ever decide to do it again.

fieldOfView, 2019-03-05:
Let's hope having a second beta is the start of a good habit.

Ghostkeeper, 2019-03-05:
The second beta was intended to be "just this once" because of the UI change and the cloud connectivity thing. The latter was also the reason for the second beta taking so long to be released by the way. They were having some trouble implementing the firmware update that we depended on. But I agree that it's a good habit to do for big releases like a major version increment.