Ultimaker/Cura #4525: Extra skin lines should have same width as Top surface skin line width Cura 3.5.0

printingotb, 2018-10-09:
**Application Version**
3.5.0

**Platform**
Win7

**Printer**
Wanhao Di3

**Steps to Reproduce**
Set overall line width to 0.4
Increase top/bottom line widths to 0.8
Enable at least 1 Top surface skin layers
Make sure Top surface skin line width is same as overall line width.
Set Extra skin wall count to at least 1
Also see attached project file.

**Actual Results**
The extra skin line will receive 0.8mm line width while the wall lines and top surface lines receives 0.4mm line widths. The single (or multiple) wider skin lines is less visually pleasing.

**Expected results**
The extra skin lines should have the same width as Top surface skin lines on the layers using top surface skin.

**Additional Information**
In this specific case, I would like to use many more extra skin lines to get a nicer pattern on the flat circular area. But then I would need to decrease top/bottom line widths to avoid having uglier wide lines.
I am using wider line widths where it's not visible to save print time. 
[image]
[Surface skin.zip](https://github.com/Ultimaker/Cura/files/2460125/Surface.skin.zip)

smartavionics, 2018-10-13:
Hi. I have looked into this to see if it is possible to achieve what you are asking and, I'm afraid to say, it's not an easy change to make. Changing the printed width of that extra skin line is straightforward. The hard bit is that where the outline of those extra skin lines are calculated (and the outline of the skin region that lies inside of the extra skin lines) it's not known whether the skin is a top surface or not so it cannot determine which line width to use for the extra skin lines at that point.

Here's an image that shows the problem:

[image]

The extra skin line has the width you desire but it has gaps around it because the part of Cura that does the spacing of that line and the skin region inside of it doesn't know that the line width it should be using to determine the spacing should be the top layer skin line width rather than the normal skin line width.

printingotb, 2018-10-13:
I suspected as much. Thanks for looking into it and taking your time to explain. Appreciate it.

closed 2026-09-06 by GregValiant