Ultimaker/Cura #4526: Explicitly set printing nozzle after start code

AbeFM, 2018-10-09:
**Application Version**
3.5.0

**Platform**
win64

**Printer**
Dual Extruder Custom

**Steps to Reproduce**
Have start g-code, print something

**Actual Results**
If the start-up g-code leaves T1 active, the first layer will print with T1, if it's T0, it's T0

**Expected results**
Material should print with the nozzle shown in the preview

**Additional Information**
I add a line:
;T0; uncomment this is first printing is with T0
;T1; uncomment this is first printing is with T1

It seems like the kind of thing a computer could do.

AbeFM, 2018-10-12:
Obviously no one is "responsible" to make my printer work.

But if there's a hole in the logic that I've exposed, I'd like to see CURA fixed.

Am I to assume you're asking for a peyote? It makes sense you can't see one as I haven't posted one yet. I would be happy to if you think it will help.

smartavionics, 2018-10-12:
A Peyote? Hell, yeah, now you're talking!

smartavionics, 2018-10-12:
aka autocorrect rides again?

AbeFM, 2018-10-12:
Always learning. I guess this:
[image] is what "previews" look like.

In more seriousness, I'm just trying to help. If there's something I can provide, let me know and I will do my best to provide it.

Ghostkeeper, 2018-12-19:
We will not solve this. The entire reason for existing of the start g-code is to prepare the printer for printing. That means that you might have to prime the nozzle, wipe it off if your printer has a wiping brush, perhaps heat up the extruders, sacrifice some virgins to the god of PLA, and to make sure that the correct tool is active.

If Cura would always make it select the first tool to print with, you would have some printers that will always at the start of a print move to their tool switching location and make some useless move and then move back (and in the process ooze some material) because they blindly follow the g-code regardless of whether T0 was already active or not. That's why we make that sort of thing configurable: Some printers need it, some don't.

closed 2018-12-19 by Ghostkeeper

AbeFM, 2018-12-19:
I would submit that [first printing nozzle] could be a variable used in start up code.

Having to hand edit start up code because you're using one of the other nozzle seems... Incongruous with the rest of the program being GUI based.

I've got things working, so it's not to help me. I just wanted to record my logic for posterity. Thanks for explicitly acknowledging my topic.

fieldOfView, 2018-12-19:
Try ```T{initial_extruder_nr}``` in your start gcode.

AbeFM, 2018-12-19:
Thanks, I will!