Ultimaker/Cura #4530: [3.5] edited file name is not saved.

Donner-Party, 2018-10-10:
**Application Version**
3.5

**Platform**
Windows 10 x64 build 1809

**Printer**
Geeetech i3W

**Steps to Reproduce**
load an STL
edit the file name
save the file
the name of the saved file does not match the edited file name.

**Actual Results**
the saved file name is the same as before I edited it.

**Expected results**
The name of the saved file should be the same as the file name after I edited it.
I just "upgraded" to 3.5 and did not have this problem in 3.3.

**Additional Information**

[image]

[CFFFP_Center Finder (1).zip](https://github.com/Ultimaker/Cura/files/2463123/CFFFP_Center.Finder.1.zip)

[stderr.zip](https://github.com/Ultimaker/Cura/files/2463129/stderr.zip)

LipuFei, 2018-10-10:
@diegopradogesto Could you try this on Windows? I could not reproduce this on Linux.

diegopradogesto, 2018-10-10:
No, I can't reproduce it. I tried several things and all of them worked for me. Let's wait until somebody else reported the same issue. Otherwise we will close it.

LipuFei, 2018-10-10:
Hi @Donner-Party , wonder if this is still a problem for you?

Donner-Party, 2018-10-10:
Yes, still a problem. Should I remove Cura and re-install?

LipuFei, 2018-10-11:
@Donner-Party Yes, please try that. And we have a final 3.5 release (not BETA). Can you check if you are using the final release? Thanks

pete-ladell, 2018-10-11:
I've had this problem since at least v3.3 using win 7. 1st showed when Cura introduced printer models into the file name to be saved. (UM2_FileName) and still with v3.5 Just tried it with win 10 and v 3.2.1 works as it should - saves as edited. & again with 3.5 seems to work as above.
Another difference between 3.2.1 & 3.5  on win 7 I get "Save to File" or "Save to Removable Drive" on win 10 I get "Prepare" before Save to ...

ikout, 2018-10-12:
I have had this issue as well with the last versions of Cura and it still happens at 3.5 at MacOS...

[image]

ikout, 2018-10-12:
I think I found the "**bug**"... If the STL File name has spaces, ie. "Bed Level Testing" and I try to rename it to "Bed Level Testing_v2" the name will be shown at the "filename" field but when I press save due to the spaces at the filename it will save it as the original name....

If the filename has no spaces ie. "Bed_Level_Testing" and I rename it to "Bed_Level_Testing_v5" then it will be saved with the "new" file name:

[image]

Either handle the filenames with spaces better or put a message that the file cannot be renamed due to spaces at the filename.