Ultimaker/Cura #4533: [Feat.REQ] "Support only" Coasting

swatti87, 2018-10-10:
Coasting that only aplies to supports and its components. Entirely separate from "normal" coasting.

I have small overhangs that droop if not supported, they are very close to a nearby wall and supports tend to carry over a bit of filament leaving a "fish fin" like leftover attached to the wall.

Suppose i used "a bit too large value" of coasting in each support, they would be cut short and no filament would be carried over to the wall.

"Just an idea"

swatti87, 2018-10-11:
Maybe an easier thing to implement: retract after support with sub-options to double the ret-distance, z-hop and run backwards to wipe the nozzle. Each separate option.

Ghostkeeper, 2018-12-18:
This is already implemented. Please enable the Enable Bridge Settings setting and adjust the Bridge Wall Coasting setting as needed.

It only works for walls though. Implementing this also for skin is a technical problem due to the way in which skin lines are generated.

closed 2018-12-18 by Ghostkeeper