Ultimaker/Cura #4534: [3.5] Turn off heat bed after the first layer

Liger0, 2018-10-10:
**Application Version**
3.5

**Platform**
Windows 10 x64

**Printer**
Cr-10s

**Steps to Reproduce**
Set a first layer target bed temperature as you want ( I do 45°C), and both preset and "normal" bed temperature at 0°C

**Actual Results**
The bed temperature stays at 45°C for all the print

**Expected results**
After the first layer the bed should cool down

Liger0, 2018-11-18:
The workaround I am using is to set the bed temperature at 1°C instead of 0°C.

nallath, 2019-01-07:
This is probably a firmware issue of the printer. But since there is a pretty easy work around for this, this can be closed.

Liger0, 2019-01-25:
> 
> 
> This is probably a firmware issue of the printer. But since there is a pretty easy work around for this, this can be closed.

It isn't.
With other slicers it works just fine, only cura has this issue.
When setting a temperature of 0°C the slicer should send the gcode command to turn off the bed heater.
It clearly is a Cura issue, so please consider opening this ticket again. Thanks.

LipuFei, 2019-05-03:
Hi @Liger0 , your workaround with 1°C instead of 0°C should work. 0°C is designed to be ignored in CuraEngine. You can also find this information in help tooltip of the bed temperature setting:

> The temperature used for the heated build plate. If this is 0, the bed temperature will not be adjusted

LipuFei, 2019-05-03:
Hi @Liger0 , we have discussed and the current behavior (do nothing with 0°C) has been there for a long time. We don't know if other printers also depend on this behavior so simply changing it may break things. So, we decided not to do thing and sorry that you have to use the workaround.

closed 2019-05-03 by LipuFei