Ultimaker/Cura #4536: [3.5] Combin "not in the coating" and "in the  infill" are both rpesent

Liger0, 2018-10-10:
**Application Version**
3.5

**Platform**
Windows 10 x64

**Printer**
Cr-10s

**Steps to Reproduce**
Check the combing mode options.

**Actual Results**
You can select either "not in the coating" and "in the infill"

**Expected results**
Since even in the combing description it says the "In the infill" is the option which has made obsolete the old "not in the coating" option, the latter should not be present.

**Additional Information**
[image]

Liger0, 2018-10-10:
> 
> 
> Hello. The old implementation has been retained so that if people do not like the new implementation they can always go back to what they had before. Sorry, I can only work in English so if your issue is with the language then someone else will have to answer.

The description says they are the same settings, just in old cura versions it was called "not in the coating" while now it is called "in the infill". If they are the same, why keeping both?

smartavionics, 2018-10-10:
They are not the same, they behave very differently. Perhaps the translation is poor?

Liger0, 2018-10-10:
> 
> 
> They are not the same, they behave very differently. Perhaps the translation is poor?

Seems so then.
It literally translates into "Note that the "In infill" settings just acts like "not on the coats" of the old cura versions." 
So I can't catch the differences.

Ghostkeeper, 2018-12-13:
The English options are "Not in Skin" (meaning, don't avoid collisions when moving inside the skin but retract instead), or "Within Infill" (meaning, only avoid collisions when moving through the infill).

What should the Italian translations be for those options, you think? The word "skin" seems to be translated as "coat" in your language so we should keep that consistent.

Ghostkeeper, 2018-12-13:
I think the confusion comes from the last sentence of the description:

    Note that the 'Within Infill' option behaves exactly like the 'Not in Skin' option in earlier Cura releases.

I think I'll remove that sentence now and consider this fixed. The translators will then also remove the sentence and then the confusion should be gone. Information about an old release of Cura will become gradually less relevant over time.

commit 3d4da94b (2018-12-13): Remove obsolete description detail about old Cura versions

This setting had a detail in the description about how the setting used to behave in old Cura versions. Let's remove it since it is not relevant any more.

Fixes #4536.
closed 2018-12-13 by Ghostkeeper