Ultimaker/Cura #4537: [3.5 Final] Wrong Temp Output

CCS86, 2018-10-10:
**Application Version**
 3.5

**Platform**
 Win 10 x64

**Printer**
 Prusa i3 MK3

**Steps to Reproduce**
 Manually set all available temperatures

**Actual Results**
 Wrong temperature output in gcode

**Expected results**
 Correct temperature output in gcode

[image]

I really wish that you guys could give me a checkbox to disable all the complicated temperature settings. I have been printing for 6 years and prefer a single printing temp from start to finish. It is tiring having to set 5 different temperatures and fight the automatic calculated values.

LipuFei, 2018-10-10:
Hi @CC86 , thanks for reporting. This is the same issue as https://github.com/Ultimaker/Cura/issues/4510, which has been fixed. The fix will be included in the next release 3.5.1.

closed 2018-10-10 by LipuFei

CCS86, 2018-10-11:
Thank you. Sorry for not searching more thoroughly.