Ultimaker/Cura #4538: Skin Direction Does not change between layers

ctmakro, 2018-10-10:
**Application Version**
3.5 installer from cura website

**Platform**

windows

**Printer**

custom printer

Two identical part on the plate.
One of them is **mirrored** from the other.
Normally the skin would change direction every layer, but I got this:

layer 79 (support top interface)
[image]

layer 80 (skin
[image]

layer 81 (skin
[image]

smartavionics, 2018-10-10:
You have not attached a project so I can only guess what the settings are. Anyway, it looks like the old bridged skin detection code has decided those skin areas are bridges (i.e. not supported) and has done its best to orientate the skin lines. If you enable the experimental bridge settings and set the bridge skin support threshold to not a high value (I think the default is 50 which should be OK) then those skins should not be considered to be bridges as they are above support. I hope this helps.

ctmakro, 2018-10-10:
sorry for the lack of project file. will experiment bridge settings now, thanks for the advice.

smartavionics, 2018-10-10:
> although bridged skin detection code might have a bug here since the entire area is very well supported.

The original bridged skin detection code does not take into account the support. When the experimental bridge settings are not enabled, Cura reverts to using the original code.

Ghostkeeper, 2018-12-13:
This bug still reproduces in the current Master version.

Ghostkeeper, 2018-12-13:
CuraEngine is supposed to explicitly choose the line direction to be perpendicular to skin, but it seems to have an off-by-one issue there. You can see that even if you set the Z distance to an even multiple of the layer height (like 0.2mm in your case) then the line direction is still not perpendicular to the skin.

smartavionics, 2018-12-13:
> This bug still reproduces in the current Master version.

That's because the original bridged skin detection code is still the same. If you use the experimental bridge settings then this problem should not occur.

Ghostkeeper, 2018-12-14:
It has nothing to do with bridging, since this affects only the lines that are not adjusted for bridges (since only those are supposed to be perpendicular to the support interface lines). If there was a fix in the experimental bridge settings then we should remove it from there and implement it to work only on lines that were not adjusted for bridges.

smartavionics, 2018-12-14:
> It has nothing to do with bridging, since this affects only the lines that are not adjusted for bridges (because only those are supposed to be perpendicular to the support interface lines).

OK, well in that case, just ignore what I said above.

Ghostkeeper, 2018-12-14:
I'm sorry, @smartavionics I was wrong! I was confused because the bridging lines are in this case exactly 90 degrees rotated so I thought that the other lines were the normal skin direction. @rburema analysed this and established that it was indeed the old bridging routine that just doesn't pay attention to the direction of the supporting lines. You were right all along.

smartavionics, 2018-12-14:
OK, no problem.

Ghostkeeper, 2018-12-18:
We're planning now to remove the old bridging procedure (and this bug with it). This involves a couple of setting changes to ensure that the default behaviour is still the same as what it was until better profiles have been tuned for it:
* Enable Bridging Settings is now enabled by default and the setting removed. It'll always use the new routine.
* Bridge Skin Support Threshold to 100%. (Note that this setting is currently not functional.)
* Bridge Wall Coasting to 0%.
* Bridge Wall Speed to Outer Wall Speed. (Note that this can't have the correct speed for inner walls, but the difference is so small that we accept this change.)
* Bridge Wall Flow to 100%.
* Bridge Skin Speed to Top/Bottom Speed.
* Bridge Skin Flow to 100%.
* Bridge Has Multiple Layers disabled.

That's our plan for fixing this bug and future bugs related to bridging. It's a bit of a refactor though so it'll take some time.

GregValiant, 2024-11-15:
Is this still an issue with current Cura versions (5.8.0 and up).  Can this be closed?

closed 2024-12-09 by github-actions[bot]