Ultimaker/Cura #4539: Cura Crashes when opening files on a USB drive

jpcode32, 2018-10-10:
**Application Version**
3.5

**Platform**
Windows 10

**Printer**
Tevo Tornado and Creality ender 3

**Steps to Reproduce**
Open an STL file on a USB drive that was created in Autodesk Inventor

**Actual Results**
Cura crashes and closes

**Expected results**
Cura should have loaded the STL file

**Additional Information**
 This does not happen if I use Cura 3.4 or 15.04.6

LipuFei, 2018-10-10:
Hi @jpcode32 , this is probably the same issue as https://github.com/Ultimaker/Cura/issues/4523. There is a Cura 3.5.1 candidate build for testing purpose. It includes a fix for this problem. Could you try it? Thanks.

**Link:** https://we.tl/t-0gqXweB9KG

jpcode32, 2018-10-10:
Yes that fixed it